TomKing062/CVE-2022-38694_unlock_bootloader #121: Root后无法修改system分区

LimengzeDev, 2024-09-30:
我按照您的方法root成功后无法访问system分区(小度v20 Android12 ab分区)
请问如何解决？非常感谢！

zoomver, 2024-09-30:
Hello LimengzeDev.
I am writing via google translate.
Unfortunately, I do not know the brand of your smartphone, but from the title of the post, I realized that you want to change the contents of the "system" section.
To do this, you need to move the "system" section to the RW state.
Good luck to you.

LimengzeDev, 2024-10-01:
> Hello LimengzeDev. I am writing via google translate. Unfortunately, I do not know the brand of your smartphone, but from the title of the post, I realized that you want to change the contents of [...]

I am able to mount on every partition after root,but I still can't write or mount on system partition.It says that system partition is a rean-only partition.
What can I do next?

zoomver, 2024-10-01:
Hi.
This is exactly how things have been going since the Android 10+ version. After obtaining root rights, we can only read the sections and view their contents.
Your gadget has an Android 12 operating system, with block markings "AB". 
This means that there is a "super.img" section in the stock firmware of your device and the "system", "product", "system_ext" and other sections are included in it.
To be able to switch these sections to RW mode, I used the RO2RW module for Magisk (RO2RW-StableBeta.v3.7.2.1.zip and Magisk_v26.3(26300).apk)
This module creates a "super.img" image with RW mode. And a backup image of "super.img" in RO mode.
After creating the "super.img" RW image, I flash it through fastboot.
Through Root+Explorer_v4.12.3.apk, you can grant RW rights to the section you need.
And eventually, I get the "system" section and other important sections in RW mode. I can mount a partition and change its contents.
I am sending you the RO2RW module to install via Magisk. It is in English. Consider it as an option, maybe it will suit you.
There are several video guides on how to use it on YOUTUBE. Or I can give instructions (of course the translation may be slightly different, since the translator is Google translate)
Good luck.
[Uploading RO2RW-StableBeta.v3.7.2.1.zip…]()

zoomver, 2024-10-01:
Hello.
Yes, Magisk 27.0 should be suitable. In our forums, this version of Magisk is actively used with this module.
Super.img firmware command: fastboot flash super super.img
I have a ZTE Blade A31 smartphone. Flashed super.img in FastbootD mode. There were some nuances when getting root and the super.img firmware. But since you have already gained root access, I think there will be no problem with the super.img firmware.

LimengzeDev, 2024-10-02:
> Hello. Yes, Magisk 27.0 should be suitable. In our forums, this version of Magisk is actively used with this module. Super.img firmware command: fastboot flash super super.img I have a ZTE Blade A31 [...]

Sorry,but my phone went bootloop after flashing the image the module produced.I made a backup but it is still in my phone.I made the image by using the first option the script provided and the file is very large.Are there any solutions?Thanks!

TomKing062, 2024-10-02:
需要改么 改prop不是只要magisk prop功能么
另外，一是vbmeta要关闭verity，而是可能使用了erofs，即使关了verity也不能改

zoomver, 2024-10-02:
Hi. Yes, I had the same problems as you have now. As I said, my model had its own nuances. Now, you need to restore the stock firmware again and start over. I wrote you instructions about what I was doing with my smartphone. I posted this instruction on my forum. I wrote using Google translate, I hope that the translation will be normal. Maybe this will help you. Good luck.
[Screenshots for Termux.zip](https://github.com/user-attachments/files/17231676/Screenshots.for.Termux.zip)
[platform-tools_r30.0.5-windows.zip](https://github.com/user-attachments/files/17231881/platform-tools_r30.0.5-windows.zip)

Getting root rights.

1. Install the drivers for the phone (if not already installed). 
2. Set up the phone: -Activate the developer menu -Enable USB debugging (in the developer menu) -Enable USB configuration => file transfer (in the developer menu) 
3. Download: 
-Magisk.apk 26.3 
-Root Explorer  
-boot.img (magisk 26.3) 
-Stock vbmeta.img -platform-tools_r30.0.5-windows 
4. Put the patched boot.img and stock vbmeta.img in the platform-tools_r30.0.5-windows folder 
5. Connect the phone to the computer. 
6. Open the platform-tools_r30.0.5-windows folder, hold down Shift and right mouse button in any an empty place. In the context menu that appears, select “open command window” 
7. We flash the patched boot.img and stock vbmeta.img. To do this, connect the phone to the computer and in the command window that opens, write the following commands in turn, pressing Enter after each of them:

1.adb devices
2.adb reboot fastboot
3.fastboot devices
4.fastboot --disable-verity flash vbmeta_a vbmeta.img
5.fastboot --disable-verity flash vbmeta_b vbmeta.img
6.fastboot flash boot_a boot.img
7.fastboot flash boot_b boot.img
8.fastboot reboot recovery

In recovery, select wipe and after clearing the userdata, reboot. 
8. Setting up the phone:  
-installing magisk
-launch magisk, then it will offer to perform an extended installation. Click "OK". After 5 seconds, the phone will reboot. -install a file manager (for example, Root Explorer), check its performance.

Creating super.img RW, converting system to RW.

Note: A bug was found while working with the script. From the first or second time you use the script, the full image may not be created. Therefore, you need to try again. What needs to be installed: 
1. Magisk 26.3 
2. Module RO2RW-StableBeta.v3.7.2.1 (Install in the ”modules" tab of the magisk program)
3. Termux.apk 
What you need to do on your smartphone: 
1. Launch Termux.apk 
2. In termux, enter the command: su –c RO2RW and press: enter
3. view the Screenshots folder for Thermex.zip
4. The RO2RW_SUPER folder will appear in the /storage/emulated/0 directory. 
5. Copy the RO2RW_SUPER folder to your computer. 
6. In the RO2RW_SUPER folder, find the super-rw-sparse-fastboot-active-_b.img file and rename it to super.img (this will be our full super.img in RW mode). 
7. A file named super-backup-sparse-fastboot-active-_b.img is a file backup super.img, to roll back to normal if any error suddenly occurs in the system. Note: active-_b is an indication of the active slot in which this .img image is stitched 

Super.img (RW) firmware 

(To avoid getting a bootloop, follow the instructions) 
1. Download platform-tools_r30.0.5-windows 
2. Put the stock boot.img and vbmeta.img in the platform-tools_r30.0.5-windows folder 
3. Activate the developer menu. Enabling USB debugging. Select the USB configuration => file transfer. 
4. Connect the phone to the computer. 
5. Open the platform-tools_r30.0.5-windows folder, hold down the Shift and right mouse button in any available place. In the context menu that appears, select “open command window” 
6. Flash the stock boot.img and vbmeta.img. In the window that opens, write the commands one at a time, after each press Enter:
1.adb devices
2.adb reboot fastboot
3.adb reboot fastboot
4.fastboot flash vbmeta_a vbmeta.img
5.fastboot flash vbmeta_b vbmeta.img
6.fastboot flash boot_a boot.img
7.fastboot flash boot_b boot.img
8.fastboot reboot recovery

In recovery, select wipe and reboot. 
7. Set up the phone and follow step 3 
8. Put the boot.img + stock vbmeta.img + super.img (RW) patched via magisk into the platform-tools_r30.0.5-windows folder, before deleting the stock boot.img image 
9. We carry out point No. 5 
10. In the command window, write the commands one by one, after each press Enter:
1.adb devices
2.adb reboot fastboot
3.fastboot devices
4.fastboot --disable-verity flash vbmeta_a vbmeta.img
5.fastboot --disable-verity flash vbmeta_b vbmeta.img
6.fastboot flash boot_a boot.img
7.fastboot flash boot_b boot.img
8.fastboot flash super super.img
9.fastboot reboot recovery

In recovery, select wipe and reboot 

11. Set up the phone, install and configure magisk, install a file manager (for example, Root Explorer), check the performance of RW in the manager.

LimengzeDev, 2024-10-03:
> Hi. Yes, I had the same problems as you have now. As I said, my model had its own nuances. Now, you need to restore the stock firmware again and start over. I wrote you instructions about what I was doing with my smartphone. I posted this instruction on my forum. I wrote using Google translate, I hope that the translation will be normal. Maybe this will help you. Good luck. [Screenshots for Termux.zip](https://github.com/user-attachments/files/17231676/Screenshots.for.Termux.zip) [platform-tools_r30.0.5-windows.zip](https://github.com/user-attachments/files/17231881/platform-tools_r30.0.5-windows.zip)
> 
> Getting root rights.
> 
> 1. Install the drivers for the phone (if not already installed).
> 2. Set up the phone: -Activate the developer menu -Enable USB debugging (in the developer menu) -Enable USB configuration => file transfer (in the developer menu)
> 3. Download:
>    -Magisk.apk 26.3
>    -Root Explorer
>    -boot.img (magisk 26.3)
>    -Stock vbmeta.img
>    -platform-tools_r30.0.5-windows
> 4. Put the patched boot.img and stock vbmeta.img in the platform-tools_r30.0.5-windows folder
> 5. Connect the phone to the computer.
> 6. Open the platform-tools_r30.0.5-windows folder, hold down Shift and right mouse button in any an empty place. In the context menu that appears, select “open command window”
> 7. We flash the patched boot.img and stock vbmeta.img. To do this, connect the phone to the computer and in the command window that opens, write the following commands in turn, pressing Enter after each of them:
> 
> 1.adb devices 2.adb reboot fastboot 3.fastboot devices 4.fastboot --disable-verity flash vbmeta_a vbmeta.img 5.fastboot --disable-verity flash vbmeta_b vbmeta.img 6.fastboot flash boot_a boot.img 7.fastboot flash boot_b boot.img 8.fastboot reboot recovery
> 
> In recovery, select wipe and after clearing the userdata, reboot. 8. Setting up the phone: -installing magisk -launch magisk, then it will offer to perform an extended installation. Click "OK". After 5 seconds, the phone will reboot. -install a file manager (for example, Root Explorer), check its performance.
> 
> Creating super.img RW, converting system to RW.
> 
> Note: A bug was found while working with the script. From the first or second time you use the script, the full image may not be created. Therefore, you need to try again. What needs to be installed:
> 
> 1. Magisk 26.3
> 2. Module RO2RW-StableBeta.v3.7.2.1 (Install in the ”modules" tab of the magisk program)
> 3. Termux.apk
>    What you need to do on your smartphone:
> 4. Launch Termux.apk
> 5. In termux, enter the command: su –c RO2RW and press: enter
> 6. view the Screenshots folder for Termux.zip
> 7. The RO2RW_SUPER folder will appear in the /storage/emulated/0 directory.
> 8. Copy the RO2RW_SUPER folder to your computer.
> 9. In the RO2RW_SUPER folder, find the super-rw-sparse-fastboot-active-_b.img file and rename it to super.img (this will be our full super.img in RW mode).
> 10. A file named super-backup-sparse-fastboot-active-_b.img is a file backup super.img, to roll back to normal if any error suddenly occurs in the system. Note: active-_b is an indication of the active slot in which this .img image is stitched
> 
> Super.img (RW) firmware
> 
> (To avoid getting a bootloop, follow the instructions)
> 
> 1. Download platform-tools_r30.0.5-windows
> 2. Put the stock boot.img and vbmeta.img in the platform-tools_r30.0.5-windows folder
> 3. Activate the developer menu. Enabling USB debugging. Select the USB configuration => file transfer.
> 4. Connect the phone to the computer.
> 5. Open the platform-tools_r30.0.5-windows folder, hold down the Shift and right mouse button in any available place. In the context menu that appears, select “open command window”
> 6. Flash the stock boot.img and vbmeta.img. In the window that opens, write the commands one at a time, after each press Enter:
>    1.adb devices
>    2.adb reboot fastboot
>    3.adb reboot fastboot
>    4.fastboot flash vbmeta_a vbmeta.img
>    5.fastboot flash vbmeta_b vbmeta.img
>    6.fastboot flash boot_a boot.img
>    7.fastboot flash boot_b boot.img
>    8.fastboot reboot recovery
> 
> In recovery, select wipe and reboot. 7. Set up the phone and follow step 3 8. Put the boot.img (magisk 26.3)+ stock vbmeta.img + super.img (RW) patched via magisk into the platform-tools_r30.0.5-windows folder, before deleting the stock boot.img image 9. We carry out point No. 5 10. In the command window, write the commands one by one, after each press Enter: 1.adb devices 2.adb reboot fastboot 3.fastboot devices 4.fastboot --disable-verity flash vbmeta_a vbmeta.img 5.fastboot --disable-verity flash vbmeta_b vbmeta.img 6.fastboot flash boot_a boot.img 7.fastboot flash boot_b boot.img 8.fastboot flash super super.img 9.fastboot reboot recovery
> 
> In recovery, select wipe and reboot
> 
> 11. Set up the phone, install and configure magisk, install a file manager (for example, Root Explorer), check the performance of RW in the manager.

Thank you.Now I will backup the user data and wipe the system to try again with your method. The method on xda doesn't needs an
vbmeta image. Should I patch the images with magisk before flashing it and dump the vbmeta.img to flash in fastboot?

zoomver, 2024-10-03:
Hi.
Try using vbmeta.img, in my case it helped (you don't need to patch the stock vbmeta.img in magisk. This is done by commands, when flashing). You only need to patch to magisk boot.img
Good luck to you.

LimengzeDev, 2024-10-03:
> Hi. Try using vbmeta.img, in my case it helped (you don't need to patch the stock vbmeta.img in magisk. This is done by commands, when flashing). You only need to patch to magisk boot.img Good luck to you.

My phone doesn't has a backup of super partition.Can anyone has it share me?
我的手机没有super分区的备份，有的人能不能给我用一下？
Thank you very much.
十分感谢！

zoomver, 2024-10-03:
Hi.
What happened to you?

LimengzeDev, 2024-10-04:
> Hi. What happened to you?

I use recovery mode to wipe data,but it didn't work because it has a corrupted system.So it still bootloops.

zoomver, 2024-10-04:
Hello.
Please tell me, are you currently trying to restore your smartphone after the first experience or after using my instructions?
Only a complete flashing will help you from bootloop. 
You should have had a backup copy of system.img after the first experiment.
I will be waiting for your reply.
Good luck.

closed 2024-10-18 by TomKing062