TomKing062/CVE-2022-38694_unlock_bootloader #124: Not issue, but question

xthemed, 2024-10-06:
Does Infinix Hot 12 Pro (X668C) unlock fine? Does it erase data? Do i need to confirm it on phone when i execute it? Can i use magisk after this?

im sorry am just nervous 😃

xthemed, 2024-10-06:
Also, if i accidently brick my phone with it, how can i restore it? Can i flash the stock boot img?

TomKing062, 2024-10-18:
1 should work
2 will erase data
3 There were confirmation option in the unlocking process before, already removed
4 magisk should work
5 pac firmware can unbrick, but TECNO added auth in some devices‘ pac firmware, anyway spd_dump provides a way to r/w partitions in device, if you think spd_dump is hard to use, some paid toolbox (unlocktools/pandora etc) can help unbrick too
6 for spd_dump, command on ums9230 is `spd_dump --wait 300 exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec w boot BOOTFILE reset`

closed 2024-10-18 by TomKing062