TomKing062/CVE-2022-38694_unlock_bootloader #128: Something about my tests with unisoc

WinKarbik, 2024-10-20:
Hello, I try root my Realme C30, but still can't do this, because it's reboots on bootlogo(seems that bootloader dnt allow this). Okay, because there Android 11, boot.img should be signed with OEM key(I have tried with hovatek instruction creating new vbmeta.img, but nothing), I have tried flash C33 android 12(Someone have booted after flashing PAC android 12 from realme c33 and have booted and magisk work!), now seems android 12 can allow boot unsigned boot.img, but there other trouble - phone not booted(I am now wait uart adapter to check why), okay, I have tested hardcore variant - I have almost full sources of sprd android 12(idh.code + BSP), have compiled PAC, but cant flash it via Research Download, and it not work when I flashed this via spd_dump(and there I am wait uart adapter too), now I am try find any other solution. Our device can be rooted only if we flash rooted GSI. Could you help fin other solutions? I'll be happy to cooperate with all developers
@TomKing062 Have you any idea? my config ums9230_4h10_go, realme c33 use ums9230_4h10
I will be glad to receive any answer.

P.S. Sorry for my bad English

WinKarbik, 2024-10-20:
And yeah, bootloader unlocked in all this tests

TomKing062, 2024-10-20:
(when unlocked)
a11 need a signed boot(any rsa4096 key works, not only OEM)
a12 doesn's check boot sign status at all

in #77 i think it mignt be Realme part issue(since some version works, some not)

last read [wiki](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/wiki/Magisk#step-3)

WinKarbik, 2024-10-21:
> (when unlocked) a11 need a signed boot(any rsa4096 key works, not only OEM) a12 doesn's check boot sign status at all
>
> in #77 i think it mignt be Realme part issue(since some version works, some [...]

Okay, I have tried your stuff for a11, it dnt help, still reboot on bootlogo. I have checked #77 , it works on RMX3511, because they have not only a11 and it work after A12, thats why it works.

WinKarbik, 2024-10-21:
And I can fastboot boot my boot.img, but cant flash and boot from this

TomKing062, 2024-10-21:
unisoc actually banned `fastboot boot`, there is  a fully work fb_cmd_boot(), just not registered, shame

> fastboot_register("flash:", fb_cmd_flash);
	fastboot_register("erase:", fb_cmd_erase);
	fastboot_register("boot", fb_cmd_continue);
	fastboot_register("reboot", fb_cmd_reboot);
	fastboot_register("powerdown", fb_cmd_powerdown);
	fastboot_register("continue", fb_cmd_continue);
	fastboot_register("reboot-bootloader", fb_cmd_reboot_bootloader);
	fastboot_register("reboot-fastboot", fb_cmd_reboot_userspace_fastboot);
	fastboot_register("reboot-recovery", fb_cmd_reboot_recovery);
	fastboot_register("set_active:", fb_cmd_set_active);

WinKarbik, 2024-10-22:
> unisoc actually banned `fastboot boot`

Oh, checked,you are right. Okay, then one question else
python2 avbtool add_hash_footer --image AIK/image-new.img --partition_name boot --partition_size 67108864 --key rsa4096_boot.pem --algorithm SHA256_RSA4096 --prop com.android.build.boot.fingerprint:$FINGERPRINT --prop com.android.build.boot.os_version:$OS_VERSION --salt 7A91E47F8D2CFB95DCCFF13305EE3F07EDCF83A42660A811F3724E1E8B463284
Should I change $FINGERPRINT, $OS_VERSION on my values(and where I should get it? adb shell getprop empty) or I should leave so? What about salt?

WinKarbik, 2024-10-22:
Seems that no Realme issue, I have no found any devices on Tiger T612 with a11 and root, but it's work for T606 and T616

WinKarbik, 2024-10-22:
nevermind,rooted