TomKing062/CVE-2022-38694_unlock_bootloader #130: Realme Narzo N53

SudirEbi, 2024-10-28:
It's running with Unisoc Tiger T612 (12 nm) chipset , Android 14 with September 2024 patch. Can you please give the unlock file for this. Thanks

R0rt1z2, 2024-10-28:
Consider uploading your u-boot, otherwise it’s unlikely you'll receive much feedback.

SudirEbi, 2024-10-29:
Hi, I am novice in this matter. I am assuming by u-boot you meant abl or preloader.. can you name  the files to upload.. I have the firmware (dump of spd tool)
Would be a great help to me..
Thanks

R0rt1z2, 2024-10-30:
Yeah.

SudirEbi, 2024-10-30:
[u-boot-files.zip](https://github.com/user-attachments/files/17571295/u-boot-files.zip)
Uploaded u-boot files..
Hoping a reply ...

R0rt1z2, 2024-10-31:
> [u-boot-files.zip](https://github.com/user-attachments/files/17571295/u-boot-files.zip) Uploaded u-boot files.. Hoping a reply ...

Can you also upload both FDLs (1 & 2)?

SudirEbi, 2024-11-01:
> > [u-boot-files.zip](https://github.com/user-attachments/files/17571295/u-boot-files.zip) Uploaded u-boot files.. Hoping a reply ...
> 
> Can you also upload both FDLs (1 & 2)?
[fdl_files.zip](https://github.com/user-attachments/files/17600469/fdl_files.zip)
Done.. plz specify if you need any other firmware stuff..

SudirEbi, 2025-01-19:
Plz someone help on this...stuck with locked bootloader  here..

TomKing062, 2025-02-08:
i think other Realme ums9230 tools can work

SudirEbi, 2025-02-08:
> i think other Realme ums9230 tools can work

Hello, I was able to unlock bootloader successfully. I want to convey my thanks and sorry to you. 
 
Thanks : for helping community and small users like me.
Sorry : I tried the universal_9230 patch and also realme_c3x patch earlier before your message but faced some error after multiple tries.. After reading your reply i decided to try all and this worked (https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/releases/download/1.72/ums9230_Realme_C53_RMX3760_RMX3762.zip)

I tried few times and it worked..

Thanks alot again and sorry for opening issue.

closed 2025-02-08 by SudirEbi