TomKing062/CVE-2022-38694_unlock_bootloader #132: Request for Itel A70 A665L

ivsbex, 2024-11-08:
It's running with Unisoc T603 Octa-Core chipset, platform, that is shown in AIDA64 is `sp9863a`, Android 13 with last security patch from 2024-10-05. itelOS version is `itel V13.0.0 (OS13.0-T-P198-231015)`. Can you please give the unlock file for this. Thanks.

Also there is a file for Itel Vision 3 (`sc9863a` which is similar to `sp9863a`), but, as I understand, it'll not match, potentially break my phone, etc. Am I right?

Massatriof16, 2024-11-11:
bruh a70 now cn unlock bootloader

Massatriof16, 2025-01-07:
> It's running with Unisoc T603 Octa-Core chipset, platform, that is shown in AIDA64 is `sp9863a`, Android 13 with last security patch from 2024-10-05. itelOS version is `itel V13.0.0 [...]

you can use that bro i've try it and it work

closed 2025-03-05 by TomKing062

SoftBlade7, 2026-08-19:
just tested the itel vision 3 unlock zip on an a70, works perfectly

@ani-dev21 to use the unlock tool:
1. download the itel vision 3 zip from releases, then extract it
2. reboot the phone to download mode (shutdown, then volume up+power, and when it turns on plug in a usb cable and keep pressing until it turns off and not reboot)
3. go to the extracted folder, open a cmd on it and run unlock_autopatch_9863a.bat
4. wait, press a button when it asks to until it completes
5. if it succeeds, the phone will automatically reboot to recovery and clear user data