TomKing062/CVE-2022-38694_unlock_bootloader #139: As you're a experienced unisoc dev, as I can see, can you help me?IFLYTEK AI Student pad C8 Pro(ud710)

hyj120309, 2024-12-31:
# Images of this device (1 and 2)
## FDL
[fdl.zip](https://github.com/user-attachments/files/18281453/fdl.zip)
## splloader, uboot, vbmeta, vendor, boot, recovery images (extracted using spd_dump tool)
https://hyj153.lanzouu.com/b004i346na
password:ad7d
Due to the limitation of the network disk, I used the compression of the volume, please delete the file extension after downloading

TomKing062, 2025-01-02:
新批次的ud710是fdl1过不去，这个既然能提就是旧批次吧 推测能用c6的工具解掉bl

TomKing062, 2025-01-02:
乐 用的这个fdl1还是用38691处理过的

hyj120309, 2025-01-09:
> 新批次的ud710是fdl1过不去，这个既然能提就是旧批次吧 推测能用c6的工具解掉bl

请大佬指教，我无法找到C6的解锁工具

TomKing062, 2025-01-10:
[kdxf-c6-unlock.zip](https://github.com/user-attachments/files/18379948/kdxf-c6-unlock.zip)
 旧版脚本 有丢分区风险

hyj120309, 2025-01-13:
> [kdxf-c6-unlock.zip](https://github.com/user-attachments/files/18379948/kdxf-c6-unlock.zip) 旧版脚本 有丢分区风险（反正你也备份了

`usb_send failed (0 / 1)`  
我遇到了以下错误，是fdl1过不去了吗？

hyj120309, 2025-01-13:
现在卡在这里了，听说展讯能在fastboot下解bl锁，不过我无法进入fastboot，您给我的工具也无法解锁（还给我卡深刷里了，扣了电池刷备份又回来（笑））

TomKing062, 2025-01-13:
用你发的那个fdl1覆盖fdl1-dl不就行了

TomKing062, 2025-01-13:
你这蓝奏云下不了，单独传下spl上来，明天给你改下

hyj120309, 2025-01-14:
> 你这蓝奏云下不了，单独传下spl上来，明天给你改下

[splloader.zip](https://github.com/user-attachments/files/18403464/splloader.zip)  
如果可以的话，非常感谢

TomKing062, 2025-01-14:
[spl-unlock.zip](https://github.com/user-attachments/files/18403616/spl-unlock.zip)

hyj120309, 2025-01-14:
> [spl-unlock.zip](https://github.com/user-attachments/files/18403616/spl-unlock.zip)

能用！！！感激不尽！！！

closed 2025-01-14 by TomKing062

hyj120309, 2025-01-14:
不过，我又遇到了一个问题，当我刷入GSI，平板就无限重启，而后进入recovery,显示“启用数据加密失败”(我翻译了，英语原文忘了)，提示可以重试,或者清除用户数据，两个我都进行了尝试，不过故障依旧，我修改官方包，将treble check加入/system/app，删了科大讯飞的一些东西，treble check显示设备支持treble，我有一台C6，也被我解锁了，刷了一样的包，都可以开机，C8我尝试了很多GSI，都无法开机，请问这该如何解决?如果可以的话，请帮助我，感激不尽!

TomKing062, 2025-01-14:
改vendor的fstab关加密，kdxf的android9都这个鬼样子

hyj120309, 2025-01-14:
> 改vendor的fstab关加密，kdxf的android9都这个鬼样子

我明天试试，