TomKing062/CVE-2022-38694_unlock_bootloader #141: 中兴平板设备无法解锁

DDZH-DEV, 2025-01-01:
[image]

使用https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/releases/download/1.72/ums9620_ZTE_universal.zip 无效，执行命令后有恢复出厂操作，但没有解锁，boot自己签名验证不通过

备份分区文件：[https://v2.z-y.site/backup.zip](https://v2.z-y.site/backup.zip)

TomKing062, 2025-01-02:
w200ds w201ds w202ds all in blacklist
有方案，目前无法公开，漏洞还没修，是不是不可能上报的，等几年再来

ll9999345, 2026-10-15:
您好，看到您在 issue #141 中提到 W200DS 曾有未公开方案，想请教目前公开的 unisoc_chipram_signcheck_exploit 是否覆盖这类设备？我的设备为中兴 W200DS / P720P01，MyOS13.0.8MR_W200DS_SYTY，Android 13，构建版本 20241013.111830，内核 5.4.254-android12-9-g656e0eec59bb，目前 Bootloader 仍锁定。是否已有适用于该版本的公开适配说明，以及所需原厂恢复材料的合法获取渠道？如果尚未公开或不适配，也请告知。谢谢。

TomKing062, 2026-10-15:
W200DS是哪个方案来着 工程uboot？
反正unisoc_chipram_signcheck_exploit 应该是能用的
zte大部分设备都没修signcheck这个

closed 2026-10-15 by TomKing062

zcop, 2026-10-20:
im success on temp unlock bootloader of  ZTE W202DS. Downside is always need redo on new boot or system softbrick as @TomKing062 mention that the exploit to persistant unlock is not work with T760 SoC