TomKing062/CVE-2022-38694_unlock_bootloader #144: Custom exec

Massatriof16, 2025-01-15:
Sir i have try custom exec noverify 4f48 on Itel A70(sc9863a) and make Check BAUD Fail but can use 4ee8 and same problem on Itel P55NFC(ums9230) can't use custom exec 65015f08 always BAUD fail i wanna ask how can find that addres and how to make custom exec?

TomKing062, 2025-01-20:
|         | spl not erase     | spl erased |
| ------- | ----------------- | ---------- |
| 9863a   | 4f48/4ee8         | 4ee8       |
| ums9230 | 65015f48/65015f08 | 65015f08   |

closed 2025-01-25 by TomKing062

Massatriof16, 2025-01-27:
> spl not erase	spl erased
> 9863a	4f48/4ee8	4ee8
> ums9230	65015f48/65015f08	65015f08

Sir can you make gen_fdl2-dl ?  Or i can use gen_fdl1-dl to generate fdl2-dl?