TomKing062/CVE-2022-38694_unlock_bootloader #146: 你能幫助我嗎？

GCR0202, 2025-01-29:
我想讓它與 Moto G14 相容，你能幫我做到嗎？我有關於引導程式等的基本知識，但這對我來說有點高水平......請幫助我...... 如果你能幫助我，我將非常感激，因為你的工具似乎是目前唯一可用的工具......我相信你必須「製作」相容性，根據你所說的，因為版本由 其他設備似乎不相容...我會等待

TomKing062, 2025-01-30:
try moto-e13?

GCR0202, 2025-01-30:
Does it work? Another question, how exactly do I do the procedure? I ended up getting lost in your tutorial... I asked the gpt chat to try to simplify it but it didn't turn out very well...

GCR0202, 2025-01-30:
I'm getting this error:

C:\Users\GCardoso\Desktop\ums9230_moto_e13>setlocal enabledelayedexpansion

C:\Users\GCardoso\Desktop\ums9230_moto_e13>if not exist "u-boot-spl-16k-sign.bin" (
spd_dump --wait 300 exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec r splloader r uboot e splloader e splloader_bak reset

 pause
 gen_spl-unlock splloader.bin
 if !errorlevel! EQU 0 (
rename "splloader.bin" "u-boot-spl-16k-sign.bin"
 chsize uboot.bin
 rename uboot.bin uboot_bak.bin
)
 pause
)  else (
spd_dump --wait 300 exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec e splloader e splloader_bak reset

 pause
)
branch:main, sha1:86fa3e31bc6ab9bd92eeaf48aa63bbfc7e065485
Waiting for dl_diag connection (300s)

After the connection message, it gets stuck and does not progress.

TomKing062, 2025-01-30:
> To enter U2S DIAG mode, it is "power + vol up" / "power + vol down" (most common) / "power + vol up + vol down"

Press key combination for 7-10 seconds to enter download mode

Execute onekey batch

GCR0202, 2025-01-30:
I did this, the goal is to enter download mode right? On mine it is: Power + Vol. Down. It entered download mode, but the script did not detect it... The drivers and cables are ok

GCR0202, 2025-01-30:
Forget what I said, I talked to a bunch of people and they told me that I need to do the test-point, if the Moto E13 version doesn't work, can you help me with a version for the Moto G14?

GCR0202, 2025-01-30:
I did everything right, but it seems that the e13 method is incompatible, after trying I get:  

C:\Users\GCardoso\Desktop\ums9230_moto_e13>setlocal enabledelayedexpansion

C:\Users\GCardoso\Desktop\ums9230_moto_e13>if not exist "u-boot-spl-16k-sign.bin" (
spd_dump --wait 300 exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec r splloader r uboot e splloader e splloader_bak reset

 pause
 gen_spl-unlock splloader.bin
 if !errorlevel! EQU 0 (
rename "splloader.bin" "u-boot-spl-16k-sign.bin"
 chsize uboot.bin
 rename uboot.bin uboot_bak.bin
)
 pause
)  else (
spd_dump --wait 300 exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec e splloader e splloader_bak reset

 pause
)
branch:stable, sha1:73d3b8cc84134cadefc69cc686c63077fd2a7886
Waiting for connection (300s)
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
current exec_addr is 0x65015f08
SEND fdl1-dl.bin to 0x65000800
SEND custom_exec_no_verify_65015f08.bin to 0x65015f08
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
CHANGE_BAUD FDL1 to 921600
KEEP_CHARGE FDL1
SEND fdl2-dl.bin to 0x9efffe00
FDL2: incompatible partition
EXEC FDL2
DISABLE_TRANSCODE
unexpected response (0x00fe)
Reading Partition List
unexpected response (0x00fe)
unexpected response (0x00fe)
timeout reached
"(This is a notice, not an error.) Don't continue if you see find port failed, just close and re-run this batch"

TomKing062, 2025-01-31:
> EXEC FDL2
DISABLE_TRANSCODE
unexpected response (0x00fe)
Reading Partition List
unexpected response (0x00fe)
unexpected response (0x00fe)
timeout reached

seems like fdl2 mismatch, we need a uboot/lk dump(or from update pack)

ThePedroo, 2025-02-01:
> > EXEC FDL2
> > DISABLE_TRANSCODE
> > unexpected response (0x00fe)
> > Reading Partition List
> > unexpected response (0x00fe)
> > unexpected response (0x00fe)
> > timeout reached
> 
> seems like fdl2 mismatch, we need a uboot/lk dump(or from update pack)
> 
> moto might have deleted [oem pull](https://github.com/TomKing062/python-fastboot-oem-pull) from their code, have you tried that?

I'm using an E20 with E13 zip due to the same unisoc. I'm getting the same errors. As requested by you, utilizing the command from the repository you sent, in a Ubuntu environment, leads to:

```
FastbootMessage(message=b'Oem Command Success', header=b'OKAY')
b'OKAY'
```

Should I create another issue, or keep on this one? I imagine E20 should also be supported as it has the same chipset, if so, I'm looking forward to help add support to it.

TomKing062, 2025-02-01:
i will re-patch e13 fdl2 to see

GCR0202, 2025-02-01:
> vou refazer o patch do e13 fdl2 para ver

This would bring possible compatibility with g14, correct? Should I try oem pull or wait for you to create the re-patch?

TomKing062, 2025-02-01:
ums9230_moto_e13_v2.zip uploaded to release

GCR0202, 2025-02-01:
> ums9230_moto_e13_v2.zip carregado para lançamento

C:\Users\gabri\Downloads\ums9230_moto_e13_v2>setlocal enabledelayedexpansion

C:\Users\gabri\Downloads\ums9230_moto_e13_v2>if not exist "u-boot-spl-16k-sign.bin" (
spd_dump --wait 300 exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec r splloader r uboot e splloader e splloader_bak reset

 pause
 gen_spl-unlock splloader.bin
 if !errorlevel! EQU 0 (
rename "splloader.bin" "u-boot-spl-16k-sign.bin"
 chsize uboot.bin
 rename uboot.bin uboot_bak.bin
)
 pause
)  else (
spd_dump --wait 300 exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec e splloader e splloader_bak reset

 pause
)
branch:stable, sha1:73d3b8cc84134cadefc69cc686c63077fd2a7886
Waiting for connection (300s)
CMD_CONNECT bootrom
current exec_addr is 0x65015f08
timeout reached
"(This is a notice, not an error.) Don't continue if you see find port failed, just close and re-run this batch"

ThePedroo, 2025-02-02:
Used E13 zip for E20, it now worked awesome! Thank you.

closed 2025-02-08 by TomKing062