TomKing062/CVE-2022-38694_unlock_bootloader #153: 执行adb reboot autodloader后，手机彻底变砖，还有办法吗？

hytTony, 2025-02-09:
乐视Letv L6B手机，T610
执行adb reboot autodloader后，彻底变砖，屏幕完全没反应。手机连接电脑时，电脑的Ports(COM & LPT)下面有SPRD U2S Diag(com3)
还有没有办法救？
谢谢！

KawaiiSparkle, 2025-02-11:
拿spddump试试，有刷机包就提取下fdl1/2，没有就去qutick102,ysepan.com找相同芯片（T610）的fdl。加载地址可以在[本项目Wiki](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/wiki/AddSupportToModel)中找到。
你应该是把splloader格了导致的，你要救砖得把splloader（你应该预先备份好了的）刷到splloader_bak和splloader分区（前面一个分区一定要刷，后面这个能刷就刷，刷不了第二个也无伤大雅），然后按键方式强制重启设备（方式是同时长按关机和挨关机最近的一个音量键）

hytTony, 2025-02-12:
好的，我试试，谢谢！

KawaiiSparkle, 2025-02-15:
如果成功了记得关闭该issue（）

hytTony, 2025-02-16:
> 如果成功了记得关闭该issue（）

好的，在查资料学习，还没开始做

hytTony, 2025-02-21:
> 拿spddump试试，有刷机包就提取下fdl1/2，没有就去qutick102,ysepan.com找相同芯片（T610）的fdl。加载地址可以在[本项目Wiki](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/wiki/AddSupportToModel)中找到。 [...]

大佬，有几个问题再详细指导一下
1、乐视Y1Pro+(Letv L6B)，T610手机，fdl1/2是用下图这个吗？
       
[image]

2、splloader我没有备份，在您网上找到海信A5Pro cc和海信A7的splloader(u-boot-spl-16k-sign.zip)，能用在我这（T610）吗？不能的话，哪里能找到？

3、把splloader(u-boot-spl-16k-sign.zip)复制到相应目录后，是执行这个命令吗？
spd_dump exec_addr 0x3ee8 fdl fdl1-dl.bin 0x5500 fdl fdl2-dl.bin 0x9efffe00 exec r boot w splloader u-boot-spl-16k-sign.bin w uboot uboot_bak.bin w misc misc-wipe.bin reset

非常感谢！

KawaiiSparkle, 2025-02-21:
spl好像不能通刷（）但是你单纯autodloader的话应该不会清除splloader_bak，所以直接读取splloader_bak分区然后刷回splloader就能开机了

hytTony, 2025-02-21:
好的，谢谢。spd_dump命令不熟，我看一下怎么读取splloader_bak分区，我试试。

TomKing062, 2025-02-22:
```
/**
 * If the SPL is erased, the romcode will enter the download process.
 * Notice: SPL must download.
 */
int erase_spl_enter_download_mode(void)
{
	stop_watchdog();
	if (0 != common_raw_erase("splloader", (uint64_t)0, (uint64_t)0)) {
		debugf("erase partition splloader fail!\n");
		return -1;
	} else {
		if (0 != common_raw_erase("splloader_bak", (uint64_t)0, (uint64_t)0)) {
			debugf("erase partition splloader_bak fail!\n");
			return -2;
		} else {
			debugf("erase partition splloader and splloader_bak ok\n");
			reboot_devices(CMD_NORMAL_MODE);
			while(1);
		}
	}

	return 0;
}
```
spl and spl_bak are erased

hytTony, 2025-02-22:
Ok, thank you. So i must get it from another same cellphone?  or somewhere can download it?

hytTony, 2025-02-22:
If I get the SPLLoader.bin file from another same cellphone, how to flash it to SPLLoader and SPLLoader_bak partitions? I am not sure if correct, could you please show me the full commands?

hytTony, 2025-02-24:
麻烦帮我看一下，读取和写入SPLLoader和SPLLoader_bak，用下面的命令对吗？谢谢！

#####读取splloader和splloader_bak#####
spd_dump read_part splloader 0 1m spl.bin
spd_dump read_part splloader_bak 0 1m spl_b.bin

#####写入splloader和splloader_bak#####
spd_dump write_part splloader spl.bin
spd_dump write_part splloader_bak spl_b.bin

TomKing062, 2025-02-25:
spd_dump exec_addr 0x3ee8 fdl fdl1-dl.bin 0x5500 fdl fdl2-dl.bin 0x9efffe00 exec read_part splloader 0 1m spl.bin reset

spd_dump exec_addr 0x3ee8 fdl fdl1-dl.bin 0x5500 fdl fdl2-dl.bin 0x9efffe00 exec write_part splloader spl.bin reset

hytTony, 2025-02-25:
好的，谢谢！等我拿到另一部一样的手机，我试试。

hytTony, 2025-02-25:
还有一个问题要请教，这款手机按电源键+音量键上（或下），或三个键都按，都进入不了下载模式。上次就是尝试进入下载模式，才执行了adb reboot autoloader。
有什么办法能在不删除SPLLoader的情况下进入下载模式？或者在非下载模式怎么读取SPLLoader？
谢谢！

TomKing062, 2025-02-25:
安卓11+还是9/10
11+可以在bootloader下提取

hytTony, 2025-02-28:
拿到新手机了，观察到这个现象：
手机关机状态下，连接电脑时，电脑的Ports(COM & LPT)下面短暂有SPRD U2S Diag(com3)，一会就消失，然后手机显示充电（电量百分比）。
是不是意味着有办法可以进入下载模式（不执行adb reboot autoloader)？应该怎能做？
麻烦再指导一下，能用下载模式的话，先用下载模式。实在不行，我再用python-fastboot-oem-pull的办法（因为看文档说明，其提取出的splloader内容好像是输出在窗口里，不知道直接复制粘贴行不行）。
谢谢！

TomKing062, 2025-02-28:
不能，关机端口(boot_diag)跳转down_diag也会擦除splloader
oem pull输出在文件里的

hytTony, 2025-02-28:
好的，我试试

hytTony, 2025-02-28:
[image]

执行python main.py oem pull partition时，报了这个错，是什么原因？

TomKing062, 2025-02-28:
命令被砍了()也可能是版本旧还没加pull()

hytTony, 2025-02-28:
还有什么办法吗？ 用OTG线换数据线，关机状态下连接电脑能进入下载模式吗？

hytTony, 2025-03-01:
又试了一下，发现连接电脑的情况下，按音量下键，Ports(COM & LPT)下面的SPRD U2S Diag(com3)能出来一会。估计在这期间再删除SPLLoader能让它保持下载模式。
麻烦再指导一下，是不是我提取出来的SPLLoader不对？应该怎么做？

hytTony, 2025-03-01:
这是提取出来的splloader，
通过网盘分享的文件：spl_b.rar
链接: https://pan.baidu.com/s/1E4fkmRFzFLpRp26t_NrhqQ?pwd=9q6s 提取码: 9q6s

TomKing062, 2025-03-01:
看不出来问题，我传个处理过的spl你试试

[spl-38691.zip](https://github.com/user-attachments/files/19038759/spl-38691.zip)

hytTony, 2025-03-02:
试了，先删除splloader，再写入spl-38691.bin，问题还是一样（SPRD U2S Diag消失，但开不了机）。
如果splloader没问题，还可能是什么地方的问题？

TomKing062, 2025-03-02:
小问题：就是版本问题（但我没发现）
大问题：你的新机旧机存在硬件混用 所以文件不通用

closed 2025-03-02 by TomKing062

hytTony, 2025-03-02:
救回来了，再从好的手机提取uboot写入就好了，可能是执行unlock_autopatch_512.bat时把uboot给覆盖了，而uboot跟我的手机不兼容。