TomKing062/CVE-2022-38694_unlock_bootloader #156: Disable secure boot on ums9820_2h10

davigamer987, 2025-02-15:
I'm trying to flash a custom vbmeta image (to allow for a custom super image) on an RG 556, after unlocking I'm only able to flash custom DTBOs using this [repo](https://github.com/TheGammaSqueeze/Bootloader_Unlock_Anbernic_T820) (These DTBO do not have to be signed with any particular key, i checked) Is there something I'm missing here? do i need to patch FDL2 (U-Boot) to disable the checks on these individual partitions? i also tried using spd_dump verity 0 but it didn't allow the device to boot

Command being used to flash vbmeta:

`spd_dump exec_addr 0x65012f48 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0xb4fffe00 exec write_part vbmeta_a vbmeta.img`

Output of spd_dump verity 0 (Using latest stable spd_dump)

Any help is appreciated, thanks!

```
spd_dump exec_addr 0x65012f48 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0xb4fffe00 exec verity 0
branch:main, sha1:8fd885f69ea9be354e0bbe89180b402803a7e0e5
Waiting for dl_diag connection (30s)
Successfully connected to port: 10
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
current exec_addr is 0x65012f48
SEND fdl1-dl.bin to 0x65000800
SEND custom_exec_no_verify_65012f48.bin to 0x65012f48
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
CHANGE_BAUD FDL1 to 921600
KEEP_CHARGE FDL1
SEND fdl2-dl.bin to 0xb4fffe00
FDL2: incompatible partition
EXEC FDL2
DISABLE_TRANSCODE
ENABLE_WRITE_RAW_DATA
Reading Partition List
[=====                                   ] 12.5%
[==========                              ] 25.0%
[===============                         ] 37.5%
[====================                    ] 50.0%
[=========================               ] 62.5%
[==============================          ] 75.0%
[===================================     ] 87.5%
[========================================] 100.0%
Read Part Done: user_partition+0x0, target: 0x8000, read: 0x8000
  0                            splloader     256KB
  1                               prodnv      64MB
  2                             miscdata       1MB
  3                                 misc       1MB
  4                            trustos_a       6MB
  5                            trustos_b       6MB
  6                                sml_a       1MB
  7                                sml_b       1MB
  8                              uboot_a       3MB
  9                              uboot_b       3MB
 10                            uboot_log      16MB
 11                                 logo       8MB
 12                            fbootlogo       8MB
 13                          nr_fixnv1_a      50MB
 14                          nr_fixnv1_b      50MB
 15                          nr_fixnv2_a      50MB
 16                          nr_fixnv2_b      50MB
 17                        nr_runtimenv1      55MB
 18                        nr_runtimenv2      55MB
 19                              persist       2MB
 20                           nr_modem_a      50MB
 21                           nr_modem_b      50MB
 22                         nr_deltanv_a       1MB
 23                         nr_deltanv_b       1MB
 24                             nr_phy_a      50MB
 25                             nr_phy_b      50MB
 26                            l_agdsp_a       6MB
 27                            l_agdsp_b       6MB
 28                             pm_sys_a       1MB
 29                             pm_sys_b       1MB
 30                             ch_sys_a      16MB
 31                             ch_sys_b      16MB
 32                             teecfg_a       1MB
 33                             teecfg_b       1MB
 34                         hypervsior_a      10MB
 35                         hypervsior_b      10MB
 36                               boot_a      64MB
 37                               boot_b      64MB
 38                        vendor_boot_a     100MB
 39                        vendor_boot_b     100MB
 40                          init_boot_a       8MB
 41                          init_boot_b       8MB
 42                                dtb_a       8MB
 43                                dtb_b       8MB
 44                               dtbo_a       8MB
 45                               dtbo_b       8MB
 46                                super    5600MB
 47                                cache      64MB
 48                             blackbox     500MB
 49                             vbmeta_a       1MB
 50                             vbmeta_b       1MB
 51                             metadata      64MB
 52                            sysdumpdb      10MB
 53                      vbmeta_system_a       1MB
 54                      vbmeta_system_b       1MB
 55                      vbmeta_vendor_a       1MB
 56                      vbmeta_vendor_b       1MB
 57                  vbmeta_system_ext_a       1MB
 58                  vbmeta_system_ext_b       1MB
 59                     vbmeta_product_a       1MB
 60                     vbmeta_product_b       1MB
 61                         vbmeta_odm_a       1MB
 62                         vbmeta_odm_b       1MB
 63                         avbmeta_rs_a       1MB
 64                         avbmeta_rs_b       1MB
 65                         common_rs1_a       8MB
 66                         common_rs1_b       8MB
 67                         common_rs2_a      16MB
 68                         common_rs2_b      16MB
 69                                ise_a       4MB
 70                                ise_b       4MB
 71                              isedata      34MB
 72                               calinv      15MB
 73                             userdata  114596MB
standard gpt table saved to pgpt.bin
skip saving sprd partition list packet
partition list saved to partition_1739615992.xml
Total number of partitions: 73
Storage is ufs
Device is using slot a
Warning: disable dm-verity needs a write-verification-disabled FDL2
unexpected response (0x00d2)
Read Part Done: vbmeta+0x0, target: 0x100000, read: 0x0
[==                                      ] 6.2%
[====                                    ] 12.5%
[=======                                 ] 18.7%
[=========                               ] 24.9%
[============                            ] 31.1%
[==============                          ] 37.4%
[=================                       ] 43.6%
[===================                     ] 49.8%
[======================                  ] 56.0%
[========================                ] 62.3%
[===========================             ] 68.5%
[=============================           ] 74.7%
[================================        ] 80.9%
[==================================      ] 87.2%
[=====================================   ] 93.4%
[======================================= ] 99.6%
[========================================] 100.0%
Read Part Done: vbmeta_a+0x0, target: 0x100000, read: 0x100000
[==                                      ] 6.2%
[====                                    ] 12.5%
[=======                                 ] 18.7%
[=========                               ] 24.9%
[============                            ] 31.1%
[==============                          ] 37.4%
[=================                       ] 43.6%
[===================                     ] 49.8%
[======================                  ] 56.0%
[========================                ] 62.3%
[===========================             ] 68.5%
[=============================           ] 74.7%
[================================        ] 80.9%
[==================================      ] 87.2%
[=====================================   ] 93.4%
[======================================= ] 99.6%
[========================================] 100.0%
Read Part Done: vbmeta_b+0x0, target: 0x100000, read: 0x100000
File vbmeta.bin does not exist, skipping.
file size : 0x100000
[========================================] 100.0%
Write Part Done: vbmeta_a, target: 0x100000, written: 0x100000
file size : 0x100000
[========================================] 100.0%
Write Part Done: vbmeta_b, target: 0x100000, written: 0x100000
```

TomKing062, 2025-02-15:
`verity 0` is for stock vbmeta
shince you don't have vbmeya.pem ,resign vbmeta won't boot

davigamer987, 2025-02-15:
> `verity 0` is for stock vbmeta shince you don't have vbmeya.pem ,resign vbmeta won't boot

I see, why does the custom dtbo work with the stock vbmeta then? It's signed with a brand new key. Should I try verity 0 on the stock vbmeta instead?

davigamer987, 2025-02-15:
Seems like after running verity 0 with the stock vbmeta it starts accepting unsigned images

closed 2025-02-15 by davigamer987