TomKing062/CVE-2022-38694_unlock_bootloader #159: 希望尝试加入对 HMD Pulse 的支持

HikariCalyx, 2025-02-23:
处理器：UMS9230

[lgd-1520.zip](https://github.com/user-attachments/files/18928107/lgd-1520.zip)

HikariCalyx, 2025-02-23:
```
spd_dump fdl fdl1-sign.bin 0x65000800 fdl lk-fdl2-sign.bin 0x9efffe00
branch:stable, sha1:73d3b8cc84134cadefc69cc686c63077fd2a7886
Waiting for connection (30s)
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND fdl1-sign.bin to 0x65000800
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
CHANGE_BAUD FDL1 to 921600
KEEP_CHARGE FDL1
SEND lk-fdl2-sign.bin to 0x9efffe00
FDL1 >repartition Legend.xml
[0] prodnv, 64
[1] elabel, 8
[2] protect_data, 8
[3] miscdata, 1
[4] misc, 1
[5] trustos_a, 6
[6] trustos_b, 6
[7] sml_a, 1
[8] sml_b, 1
[9] uboot_a, 3
[10] uboot_b, 3
[11] uboot_log, 16
[12] logo_a, 8
[13] logo_b, 8
[14] fbootlogo, 8
[15] l_fixnv1_a, 2
[16] l_fixnv2_a, 2
[17] l_fixnv1_b, 2
[18] l_fixnv2_b, 2
[19] l_runtimenv1, 2
[20] l_runtimenv2, 2
[21] persist, 2
[22] l_modem_a, 25
[23] l_modem_b, 25
[24] l_deltanv_a, 1
[25] l_deltanv_b, 1
[26] l_gdsp_a, 10
[27] l_gdsp_b, 10
[28] l_ldsp_a, 20
[29] l_ldsp_b, 20
[30] l_agdsp_a, 6
[31] l_agdsp_b, 6
[32] pm_sys_a, 1
[33] pm_sys_b, 1
[34] teecfg_a, 1
[35] teecfg_b, 1
[36] hypervsior_a, 10
[37] hypervsior_b, 10
[38] boot_a, 64
[39] boot_b, 64
[40] vendor_boot_a, 100
[41] vendor_boot_b, 100
[42] init_boot_a, 8
[43] init_boot_b, 8
[44] dtb_a, 8
[45] dtb_b, 8
[46] dtbo_a, 8
[47] dtbo_b, 8
[48] super, 5300
[49] cache, 64
[50] blackbox, 500
[51] vbmeta_a, 1
[52] vbmeta_b, 1
[53] metadata, 64
[54] sysdumpdb, 10
[55] vbmeta_system_a, 1
[56] vbmeta_system_b, 1
[57] vbmeta_vendor_a, 1
[58] vbmeta_vendor_b, 1
[59] vbmeta_system_ext_a, 1
[60] vbmeta_system_ext_b, 1
[61] vbmeta_product_a, 1
[62] vbmeta_product_b, 1
[63] vbmeta_odm_a, 1
[64] vbmeta_odm_b, 1
[65] avbmeta_rs_a, 1
[66] avbmeta_rs_b, 1
[67] common_rs1_a, 8
[68] common_rs1_b, 8
[69] common_rs2_a, 16
[70] common_rs2_b, 16
[71] calinv, 2
[72] userdata, -1
unexpected response (0x008b)
FDL1 >w uboot_a lk-sign.bin
unexpected response (0x008b)
unexpected response (0x008b)
part not exist
FDL1 >

```

TomKing062, 2025-02-25:
[lgd-1520-fdl1-boot.zip](https://github.com/user-attachments/files/18952290/lgd-1520-fdl1-boot.zip)

closed 2025-02-25 by TomKing062

HikariCalyx, 2025-02-25:
工程机上验证可行。零售机待测试。

HikariCalyx, 2025-02-25:
[ums9230_HMD_Pulse_series_by@HikariCalyxTech.zip](https://github.com/user-attachments/files/18953512/ums9230_HMD_Pulse_series_by%40HikariCalyxTech.zip)

HikariCalyx, 2025-03-12:
目前接到一台因为使用 UnlockTool 提供的通用 FDL 导致整个 UFS 被清的肯尼亚定制版 HMD Pulse，被称作 Mkopa X2。
不论是使用里面附带的 fdl2-dl 还是固件内置的 lk-fdl2-sign 注入进去之后均无法正确分区，返回 Unexpected Response (0x0084)。