TomKing062/CVE-2022-38694_unlock_bootloader #160: Moto E13 FDL1 and FDL2 hard brick on Moto G14

Maiky-xd, 2025-03-02:
Hello excuse my bad English, I was trying the files to test on the Moto E13 device and they work perfectly, the Moto G14 device has the same processor but when using it on this one the device does not turn on anymore, this is due to the difference in emmc and ufs memory, I tried to create the fdl files for the Moto G14 but I have not been successful, I used the fdl1-sign.bin and lk-fdl2-sign.bin files that are in the firmware but I always get the error code unexpected response (0x00fe), according to my theories what happens is that when sending the emmc files of the Moto E13 model the UFS memory of the Moto G14 proceeds with a reconfiguration, causing a complete cleaning of the UFS memory and losing all the data in it, I have managed to turn on the Moto G14 using Pandora tool software and I continue testing but it always ends up formatting the UFS memory, I attach the original FDL1 and FDL2 extracted from the firmware, do you have any idea on how could these 2 files be correctly modified to add support for this device? Doing this would add support to more UFS devices such as the Moto G04, Moto G04s and Moto E14 that use the same processor and memory. I offer my help to test on these UFS devices.

[https://drive.google.com/file/d/1dhOekv1GFMYcLMxLJnNPzg7T2DnoheEA/view?usp=sharing]
Original FDL1 and FDL2
[image]
FDL1 and FDL2 from Moto E13 (dead phone)
[image]

TomKing062, 2025-03-05:
uploaded

closed 2025-03-05 by TomKing062

Maiky-xd, 2025-03-05:
> uploaded

Bro you are so amazing, it works perfect thanks i love u