TomKing062/CVE-2022-38694_unlock_bootloader #161: How to unlock bootloader in Infinix smart 8 hd

mandeep1152, 2025-03-03:
I am getting this error

C:\Users\Administrator\Desktop\New folder>if not exist "u-boot-spl-16k-sign.bin" (

spd_dump --wait 300 exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec r splloader r uboot e splloader e splloader_bak reset

pause

gen_spl-unlock splloader.bin

if !errorlevel! EQU 0 (

rename "splloader.bin" "u-boot-spl-16k-sign.bin"

chsize uboot.bin

rename uboot.bin uboot_bak.bin

)

pause

)  else (

spd_dump --wait 300 exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec e splloader e splloader_bak reset

pause

)

branch:stable, sha1:73d3b8cc84134cadefc69cc686c63077fd2a7886

Waiting for connection (300s)

usb_send failed (0 / 1)

"(This is a notice, not an error.) Don't continue if you see find port failed, just close and re-run this batch"

Press any key to continue . . .

loadfile("splloader.bin") failed

Press any key to continue . . .

Please help !

TomKing062, 2025-03-05:
since your device is Released 2023, December
version released after 202308 may have blocked this unlock method

(i thought tecno provides unlock apply on their website, isn't this device supported there?)

Massatriof16, 2025-03-05:
> since your device is Released 2023, December version released after 202308 may have blocked this unlock method
> 
> (i thought tecno provides unlock apply on their website, isn't this device supported there?)

No sir, this method still works, I just tried it 3 days ago

Massatriof16, 2025-03-05:
> since your device is Released 2023, December version released after 202308 may have blocked this unlock method
> 
> (i thought tecno provides unlock apply on their website, isn't this device supported there?)

[image]

closed 2025-03-21 by TomKing062