TomKing062/CVE-2022-38694_unlock_bootloader #163: Nokia T21 平板在 U2S 端口上无法正确引导

HikariCalyx, 2025-03-04:
最近拿到了一批 Nokia T21 平板工程机，试图进入 U2S 端口引导FDL失败，错误信息如下：
```
spd_dump --wait 32767 fdl fdl1-sign.bin 0x65000800 fdl lk-fdl2-sign.bin 0x9efffe00 exec
branch:main, sha1:8fd885f69ea9be354e0bbe89180b402803a7e0e5
Successfully connected to port: 50
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND fdl1-sign.bin to 0x65000800
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
CHANGE_BAUD FDL1 to 921600
KEEP_CHARGE FDL1
SEND lk-fdl2-sign.bin to 0x9efffe00
timeout reached
```
[agt-t612.zip](https://github.com/user-attachments/files/19072796/agt-t612.zip)

TomKing062, 2025-03-05:
same reason to #94

> char buffast[1];
	memset(buffast, 0, sizeof(buffast));
	printf("%s:enter\n", __FUNCTION__);
	if (0 != common_raw_read("miscdata", (uint64_t)sizeof(char),(uint64_t)HMD_EN_API_FASTBOOT_MODE_OFFSET, buffast)) {
		debugf("common_raw_read miscdata fial !\n");
	}else{
		if(!strncmp(buffast, "1",1)){
			debugf("hmd can not download! status=%s\n",buffast);
			return -1;
		}else{
			debugf("!!!! hmd can download! status=%s\n",buffast);
		}
	}	
#ifndef WT_COMPILE_FACTORY_VERSION
    if ( !wt_check_download_permission() ) {
	    printf("%s:The device is reset to normal ...\n", __FUNCTION__);
	    reset_to_normal(CMD_NORMAL_MODE);
		return -1;
	}
#endif

full info see https://www.hmd.com/en_int/opensource

[image]

TomKing062, 2025-03-05:
for download

[fdl-dl.zip](https://github.com/user-attachments/files/19083484/fdl-dl.zip)

closed 2025-04-01 by TomKing062