TomKing062/CVE-2022-38694_unlock_bootloader #166: usb_send failed (0 / 1) error on Moto G35 (ums9620_2h10)

Maiky-xd, 2025-03-08:
Hello, again excuse my bad English, trying to add support for Moto G35, first I tried using the files from another model with the same processor (ums9620) but the process does not start, I also tried to use the fdl files from the firmware but it is the same problem, it always says the error "usb_send failed (0 / 1)", this model works by test point, I am performing the test point correctly since the spd research tool software works perfectly, I attach a photo of the terminal and spd research tool, I am willing to collaborate to try to find the problem in this device

[image]

[image]

fdl files
[FDL Moto G35 ums9620.zip](https://github.com/user-attachments/files/19140179/FDL.Moto.G35.ums9620.zip)

Maiky-xd, 2025-03-11:
Connection works now, only need to edit the fdl files pls
[image]

TomKing062, 2025-03-21:
i think it's blocked

[image]

closed 2025-03-23 by TomKing062

gatesgsm777, 2025-04-01:
> i think it's blocked
> 
> [image]

Why don't we try to create a custom LK that doesn't relock the bootloader, like some Mediatek device devs did?

TomKing062, 2025-04-01:
at least it can't be done with public CVE now

we(a group knowing a bypass exploit) don't plan to burn(report) it yet

gatesgsm777, 2025-04-01:
So for now there is no workaround or method (that can be talked about publicly) to boot a modified lk?