TomKing062/CVE-2022-38694_unlock_bootloader #170: Qin F21 Pro+ Unlock Fail | 多亲F21 Pro+解锁失败

Luochancy, 2025-03-16:
通过adb reboot autodloader进入的download | Download via adb reboot autodloader
环境：Windows 11 | System: Windows 11
使用SPL | Use SPL
以下是Log | The following is the Log
`C:\Users\Luochancy\Desktop\11>if not exist "u-boot-spl-16k-sign.bin" (
spd_dump --wait 300 exec_addr 0x3ee8 fdl fdl1-dl.bin 0x5500 fdl fdl2-dl.bin 0x9efffe00 exec r splloader r uboot e splloader e splloader_bak reset

 pause
 gen_spl-unlock splloader.bin
 if !errorlevel! EQU 0 (
rename "splloader.bin" "u-boot-spl-16k-sign.bin"
 chsize uboot.bin
 rename uboot.bin uboot_bak.bin
)
 pause
)  else (
spd_dump --wait 300 exec_addr 0x3ee8 fdl fdl1-dl.bin 0x5500 fdl fdl2-dl.bin 0x9efffe00 exec e splloader e splloader_bak reset

 pause
)
branch:stable, sha1:73d3b8cc84134cadefc69cc686c63077fd2a7886
Waiting for connection (300s)
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
current exec_addr is 0x3ee8
SEND fdl1-dl.bin to 0x5500
SEND custom_exec_no_verify_3ee8.bin to 0x3ee8
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
CHANGE_BAUD FDL1 to 921600
KEEP_CHARGE FDL1
SEND fdl2-dl.bin to 0x9efffe00
FDL2: incompatible partition
EXEC FDL2
DISABLE_TRANSCODE
Reading Partition List
[=====                                   ] 12.5%
[==========                              ] 25.0%
[===============                         ] 37.5%
[====================                    ] 50.0%
[=========================               ] 62.5%
[==============================          ] 75.0%
[===================================     ] 87.5%
[========================================] 100.0%
dump_partition: user_partition+0x0, target: 0x8000, read: 0x8000
  0                            splloader 256KB
  1                               prodnv 10MB
  2                             miscdata 1MB
  3                                 misc 1MB
  4                            trustos_a 6MB
  5                            trustos_b 6MB
  6                                sml_a 1MB
  7                                sml_b 1MB
  8                              uboot_a 1MB
  9                              uboot_b 1MB
 10                            uboot_log 4MB
 11                                 logo 8MB
 12                            fbootlogo 8MB
 13                           l_fixnv1_a 2MB
 14                           l_fixnv2_a 2MB
 15                           l_fixnv1_b 2MB
 16                           l_fixnv2_b 2MB
 17                         l_runtimenv1 2MB
 18                         l_runtimenv2 2MB
 19                          gnssmodem_a 1MB
 20                          gnssmodem_b 1MB
 21                           wcnmodem_a 10MB
 22                           wcnmodem_b 10MB
 23                              persist 2MB
 24                            l_modem_a 25MB
 25                            l_modem_b 25MB
 26                          l_deltanv_a 1MB
 27                          l_deltanv_b 1MB
 28                             l_gdsp_a 10MB
 29                             l_gdsp_b 10MB
 30                             l_ldsp_a 20MB
 31                             l_ldsp_b 20MB
 32                            l_agdsp_a 6MB
 33                            l_agdsp_b 6MB
 34                             l_cdsp_a 1MB
 35                             l_cdsp_b 1MB
 36                             pm_sys_a 1MB
 37                             pm_sys_b 1MB
 38                             teecfg_a 1MB
 39                             teecfg_b 1MB
 40                         hypervsior_a 10MB
 41                         hypervsior_b 10MB
 42                               boot_a 64MB
 43                               boot_b 64MB
 44                        vendor_boot_a 100MB
 45                        vendor_boot_b 100MB
 46                                dtb_a 8MB
 47                                dtb_b 8MB
 48                               dtbo_a 8MB
 49                               dtbo_b 8MB
 50                                super 3100MB
 51                                cache 100MB
 52                              socko_a 75MB
 53                              socko_b 75MB
 54                              odmko_a 25MB
 55                              odmko_b 25MB
 56                             vbmeta_a 1MB
 57                             vbmeta_b 1MB
 58                             metadata 16MB
 59                            sysdumpdb 10MB
 60                      vbmeta_system_a 1MB
 61                      vbmeta_system_b 1MB
 62                  vbmeta_system_ext_a 1MB
 63                  vbmeta_system_ext_b 1MB
 64                      vbmeta_vendor_a 1MB
 65                      vbmeta_vendor_b 1MB
 66                     vbmeta_product_a 1MB
 67                     vbmeta_product_b 1MB
 68                         vbmeta_odm_a 1MB
 69                         vbmeta_odm_b 1MB
 70                         avbmeta_rs_a 1MB
 71                         avbmeta_rs_b 1MB
 72                         common_rs1_a 8MB
 73                         common_rs1_b 8MB
 74                         common_rs2_a 16MB
 75                         common_rs2_b 16MB
 76                             userdata 55556MB
standard gpt table saved to pgpt.bin
skip saving sprd partition list packet
partition list saved to partition.xml
Total number of partitions: 76
Device is using slot b
[=========                               ] 24.9%
[===================                     ] 49.8%
[=============================           ] 74.7%
[======================================= ] 99.6%
[========================================] 100.0%
dump_partition: splloader+0x0, target: 0x40000, read: 0x40000
[==                                      ] 6.2%
[====                                    ] 12.5%
[=======                                 ] 18.7%
[=========                               ] 24.9%
[============                            ] 31.1%
[==============                          ] 37.4%
[=================                       ] 43.6%
[===================                     ] 49.8%
[======================                  ] 56.0%
[========================                ] 62.3%
[===========================             ] 68.5%
[=============================           ] 74.7%
[================================        ] 80.9%
[==================================      ] 87.2%
[=====================================   ] 93.4%
[======================================= ] 99.6%
[========================================] 100.0%
dump_partition: uboot_b+0x0, target: 0x100000, read: 0x100000
"(This is a notice, not an error.) Don't continue if you see find port failed, just close and re-run this batch"`

Luochancy, 2025-03-16:
现在拔线以后砖了（
好像是adb reboot autodloader进去会清splloader？

TomKing062, 2025-03-20:
对啊 反正我这没有备份（

closed 2025-03-23 by TomKing062