TomKing062/CVE-2022-38694_unlock_bootloader #171: 希望作者能支持下中兴F50随身wifi设备，用的是ums9620(展锐T760)芯片

qxdjz, 2025-04-02:


TomKing062, 2025-04-02:
can't be done with public CVE now, ZTE mifi are in blacklist if you view wiki

we(a group knowing a bypass exploit) don't plan to burn(report) it yet

closed 2025-04-02 by TomKing062

hicode002, 2025-04-26:
> can't be done with public CVE now, ZTE mifi are in blacklist if you view wiki
> 
> we(a group knowing a bypass exploit) don't plan to burn(report) it yet

如果brom exploit没有被修复那么临时解锁bl总是可以的吧，另外如果要永久解锁bl，去看unlock的逻辑不行吗还是说zte压根没有unlock 相关的只能靠漏洞？

hicode002, 2025-04-26:
@TomKing062

TomKing062, 2025-04-26:
临时解意味着每次关机重启后连接电脑或另一台手机发送引导文件才能开机

hicode002, 2025-04-26:
> 临时解意味着每次关机重启后连接电脑或另一台手机发送引导文件才能开机

这个随身wifi本来是不是就必须每次连接电脑

hicode002, 2025-06-08:
> can't be done with public CVE now, ZTE mifi are in blacklist if you view wiki
> 
> we(a group knowing a bypass exploit) don't plan to burn(report) it yet

thanks for your tips!Now I successfully unlock f50 bootloader.I patched uboot with a nonpublic exploit.

TomKing062, 2025-06-08:
怎么觉得知道那个方法的人又变多了（