TomKing062/CVE-2022-38694_unlock_bootloader #173: Help me :(

GCR0202, 2025-04-09:
After trying to unlock the Motorola Moto G14, my device is stuck in "Download Mode" / SPD Mode. It does not turn on or start Android, But the computer still recognizes it and can send some commands. I don't know what to do.

TomKing062, 2025-04-10:
log or screenshot

GCR0202, 2025-04-11:
> log ou captura de tela

I think I no longer have the log of the first error, because I deleted the folder and extracted it again after the error. The only thing in the logs are error lines. "Call CreateFile(), fail, [ERRORCODE: 0X00000079]" "Call ClearCommError() fail, [ErrorCode: 0X000003E3] and [0x00000016]

TomKing062, 2025-04-12:
how didi you enter u2s when running unlock batch
consider as splloader lost

GCR0202, 2025-04-12:
> como você entrou no u2s ao executar o desbloqueio em lote, considere o splloader perdido

So I lost my device?? Is there a way to recover it right???

TomKing062, 2025-04-12:
btw that log is useless, only custdebug build(in github action or https://nightly.link/TomKing062/spreadtrum_flash/workflows/build/main ) give detailed log

GCR0202, 2025-04-12:
> a propósito, esse log é inútil, apenas a compilação do custdebug (na ação do github ou https://nightly.link/TomKing062/spreadtrum_flash/workflows/build/main ) fornece um log detalhado

I just don't have the log where the error occurred, but what I can say is that it loaded part of the unlock and gave an error, I believe this error corrupted something, how should I recover?

TomKing062, 2025-04-12:
1 check if there are u-boot-spl-16k/uboot_bak files in folder, though i think spl is lost
2 find part dump or pac firmware on internet
3 official service

GCR0202, 2025-04-12:
> 1 check if there are u-boot-spl-16k/uboot_bak files in folder, though i think spl is lost 2 find part dump or pac firmware on internet 3 official service

I even found the firmware PAC and tried to flash it, but I remember getting a "timeout" error, I'm not mistaken.

Technical-iReverse, 2025-04-12:
@GCR0202 Repair with Pandora.

That's because your device is using UFS memory but FDL2 built is for eMMC.

FDL2 sent is causing for your device stuck at download state.

GCR0202, 2025-04-12:
> [@GCR0202](https://github.com/GCR0202)Reparar com Pandora.
> 
> Isso ocorre porque seu dispositivo está usando memória UFS, mas a FDL2 é para eMMC.
> 
> O FDL2 enviado está fazendo com que seu dispositivo trave no estado de download.

I... couldn't/don't know how to use Pandora...

gates-pro, 2025-04-16:
Hi, did you use the folder that tomking062 made for moto e13 to try to unlock g14? if so, your partitions is gone (**e13 is emmc, g14 is ufs**). If you used the folder for G14, then maybe you just lost splloader.

Anyway I'll send you a folder with some original G14 firmware images + correct FDLs and a .bat that will flash them. (splloader and lk included) If it doesn't work, I'll make another one with more important images. Doing this (with the appropriate files) saved my E13 which was bricked for months.

GCR0202, 2025-04-16:
> Olá, você usou a pasta que o tomking062 criou para o moto e13 para tentar desbloquear o g14? Se sim, suas partições desapareceram ( **e13 é emmc, g14 é ufs** ). Se você usou a pasta para o G14, talvez tenha perdido o splloader.
> 
> De qualquer forma, vou te enviar uma pasta com algumas imagens originais do firmware G14 + FDLs corretas e um .bat que as gravará. (Splloader e lk inclusos). Se não funcionar, criarei outra com imagens mais importantes. Fazer isso (com os arquivos apropriados) salvou meu E13, que estava travado há meses.

Really??? Thank you very much, I hope it works, I'll wait🤗

TomKing062, 2025-04-16:
i tried to classify  uboot/lk files before
a strange thing is , in IDA , the uboot/lk(moto ones) all have emmc/ufs detect func in it

gates-pro, 2025-04-17:
> Really??? Thank you very much, I hope it works, I'll wait🤗

It's ready. The goal is make your phone access fastboot again.

[g14.zip](https://www.mediafire.com/file/akg6e8nn8a7ujwi/g14.zip/file)

1. Make sure these drivers (this version) are installed on your computer: [SPRD Drivers](https://androiddatahost.com/dsa6h)
2. Unzip and open the g14.zip and run flash.bat
3. Do testpoint in the device, the same way you did when you tried to unlock it.
4. If you experience failure such as device connecting but commands not starting or various errors, please submit a screenshot of the command prompt here.
5. If the device turns on at the end of the flash, it will probably restart infinitely in the logo, but it doesn't matter, you can simply hold volume down + power at the same time to enter fastboot.

By entering fastboot mode you will be able to flash the stock ROM

1. Install Motorola Drivers this time: [Motorola Drivers](https://pt-br.support.motorola.com/app/usb-drivers)
2. Download this ROM, unzip and run flash_all.bat: [FASTBOOT ROM](https://www.stockrom.net/2025/03/xt2341-1-retbr-os14-utlbs34-102-62-1.html)

If everything goes well, your device is back.

If it doesn't work, let us know here, maybe some more adjustments or repartition can be made.

GCR0202, 2025-04-17:
> > > Olá, você usou a pasta que o tomking062 criou para o moto e13 para tentar desbloquear o g14? Se sim, suas partições desapareceram ( **e13 é emmc, g14 é ufs** ). Se você usou a pasta para o G14, talvez tenha perdido o splloader.
> > > De qualquer forma, vou te enviar uma pasta com algumas imagens originais do firmware G14 + FDLs corretas e um .bat que as gravará. (Splloader e lk inclusos). Se não funcionar, criarei outra com imagens mais importantes. Fazer isso (com os arquivos apropriados) salvou meu E13, que estava travado há meses.
> > 
> > 
> > Really??? Thank you very much, I hope it works, I'll wait🤗
> 
> It's ready. The goal is make your phone access fastboot again.
> 
> [g14.zip](https://www.mediafire.com/file/akg6e8nn8a7ujwi/g14.zip/file)
> 
> 1. Make sure these drivers (this version) are installed on your computer: [SPRD Drivers](https://androiddatahost.com/dsa6h)
> 2. Unzip and open the g14.zip and run flash.bat
> 3. Do testpoint in the device and connect cable (at the same time), the same way you did when you tried to unlock it.
> 4. If you experience failure such as device connecting but commands not starting or various errors, please submit a screenshot of the command prompt here.
> 5. If the device turns on at the end of the flash, it will probably restart infinitely in the logo, but it doesn't matter, you can simply hold volume down + power at the same time to enter fastboot.
> 
> By entering fastboot mode you will be able to flash the stock ROM
> 
> 1. Install Motorola Drivers this time: [Motorola Drivers](https://pt-br.support.motorola.com/app/usb-drivers)
> 2. Download this ROM, unzip and run flash_all.bat: [FASTBOOT ROM](https://www.stockrom.net/2025/03/xt2341-1-retbr-os14-utlbs34-102-62-1.html)
> 
> If everything goes well, your device is back.
> 
> If it doesn't work, let us know here, maybe some more adjustments or repartition can be made.

Thank you very much, I will test it

GCR0202, 2025-04-17:
> > > Olá, você usou a pasta que o tomking062 criou para o moto e13 para tentar desbloquear o g14? Se sim, suas partições desapareceram ( **e13 é emmc, g14 é ufs** ). Se você usou uma pasta para o G14, talvez tenha perdido o sploader. 
> > > De qualquer forma, você enviará uma pasta com algumas imagens originais do firmware G14 + FDLs corretas e um .bat que irá gravar. (Splloader e lk inclusos). Se não funcionar, criei outra com imagens mais importantes. Fazer isso (com os arquivos protegidos) salvou meu E13, que estava travado há meses.
> > 
> > 
> > Sério??? Muito obrigado, espero que funcione, aguardo 🤗
> 
> Está pronto. O objetivo é fazer com que seu telefone acesse o fastboot novamente.
> 
> [g14.zip](https://www.mediafire.com/file/akg6e8nn8a7ujwi/g14.zip/file)
> 
> 1. Certifique-se de que estes drivers (esta versão) estejam instalados no seu computador: [Drivers SPRD](https://androiddatahost.com/dsa6h)
> 2. Descompacte e abra o g14.zip e execute flash.bat
> 3. Faça um teste no dispositivo e conecte o cabo (ao mesmo tempo), da mesma forma que você fez quando tentou desbloqueá-lo.
> 4. Se você tiver falhas, como dispositivo conectando, mas comandos não iniciando ou vários erros, envie uma captura de tela do prompt de comando aqui.
> 5. Se o dispositivo ligar no final do flash, ele provavelmente reiniciará infinitamente no logotipo, mas não importa, você pode simplesmente segurar o volume para baixo + energia ao mesmo tempo para entrar no fastboot.
> 
> Ao entrar no modo fastboot você poderá atualizar a ROM de estoque
> 
> 1. Instale os drivers da Motorola desta vez: [Drivers da Motorola](https://pt-br.support.motorola.com/app/usb-drivers)
> 2. Baixe esta ROM, descompacte e execute flash_all.bat: [FASTBOOT ROM](https://www.stockrom.net/2025/03/xt2341-1-retbr-os14-utlbs34-102-62-1.html)
> 
> Se tudo correr bem, seu dispositivo estará de volta.
> 
> Se não funcionar, avise-nos aqui, talvez seja possível fazer mais alguns ajustes ou reparticionar.

[image]

It's been on this screen for a while, should I wait longer?

gates-pro, 2025-04-17:
Você também é br?

Faz assim
[Flash.bat novo](https://www.mediafire.com/file/jlif3fd7493xne7/flash.zip/file)
Extrai do zip e copia esse flash.bat pra pasta, substituindo o atual e tenta com ele

GCR0202, 2025-04-17:
> Você também é br?
> 
> Faz assim [Flash.bat novo](https://www.mediafire.com/file/jlif3fd7493xne7/flash.zip/file) Extrai do zip e copia esse flash.bat pra pasta, substituindo o atual e tenta com ele

Cê tá zoando q vc era Br? Ksks, vou fazer o teste aquu

GCR0202, 2025-04-17:
> Você também é br?
> 
> Faz assim [Flash.bat novo](https://www.mediafire.com/file/jlif3fd7493xne7/flash.zip/file) Extrai do zip e copia esse flash.bat pra pasta, substituindo o atual e tenta com ele

Agora ele dá esse erro, literalmente n foi nem 5 segundos

[image]

gates-pro, 2025-04-17:
Fecha o Bat abre ele dnv e tenta conexão dnv (testpoint + conectar cabo)
E ver se dá o mesmo erro

GCR0202, 2025-04-17:
> Fecha o Bat abre ele dnv e tenta conexão dnv (testpoint + conectar cabo) E ver se dá o mesmo erro

Agora não houve erro, mas parou aqui:

[image]

gates-pro, 2025-04-17:
🤔 talvez seja um problema com os FDLs ou com a forma que tá escrito os comandos

Mas as suas partições tão boas ainda dá pra ver pq aparece a lista completa, se tivesse corrompida aparecia só splloader

Se quiser me chama no telegram
[t.me/gatesgsm](https://t.me/gatesgsm)

Talvez seja bom vc tentar abrir um cmd na pasta e fazer
spd_dump --wait 300 exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec
Ai vc conecta e quando descer a lista de partições vai ficar no modo interativo FDL2
Ai vc só vai dando os comandos
w splloader splloader.bin
E fazendo pra cada partição

GCR0202, 2025-04-17:
> 🤔 talvez seja um problema com os FDLs ou com a forma que tá escrito os comandos
>
> Mas as suas partições tão boas ainda dá pra ver pq aparece a lista completa, se tivesse corrompida aparecia só [...]

Pode me passar os comandos?

gates-pro, 2025-04-17:
w splloader splloader.bin
Ai da enter
w uboot uboot.bin
Ai tenta
reset
Se ligar deu certo, segura volume menos e power pra ir pro fastboot
Se não ligar, me manda o arquivo partition_.xml vou tentar fazer uma pasta com os nomes certos das partições

GCR0202, 2025-04-17:
> w splloader splloader.bin Ai da enter w uboot uboot.bin Ai tenta reset Se ligar deu certo, segura volume menos e power pra ir pro fastboot Se não ligar, abre o arquivo partition_.xml e copia e cola aqui o conteúdo dele ai eu faço um dump com os nomes certos das partições pra flashar com write_parts

Ele deu uma travada dps do uboot.bin

[image]

closed 2025-04-18 by GCR0202