TomKing062/CVE-2022-38694_unlock_bootloader #174: Device Support Request - Realme Note 60 (ums9230)

TheAabedKhan, 2025-04-10:
Hello, dev!

The ums9230_Realme_note_50 zip file is only able to factory reset my Realme Note 60 (RMX3933). I strongly believe you can help me unlock the bootloader for this phone. I will provide all the necessary files that is required for you to make the unlock script for this phone. 

I appreciate your development and support.

TomKing062, 2025-04-11:
send uboot_bak.bin(uboot/lk)

TheAabedKhan, 2025-04-11:
[uboot_bak.bin.zip](https://github.com/user-attachments/files/19705584/uboot_bak.bin.zip)

TomKing062, 2025-04-11:
blocked
[image]

hicode002, 2025-04-28:
@TomKing062  你发现的未公开的漏洞能够用于解锁这个设备吗？（是哪个启动环节的漏洞）（只是好奇问问）

TomKing062, 2025-04-28:
@hicode002 目前已知的对（不可持续）漏洞的缓解是在0x2000-0x2040的解锁flag上额外加上
厂商私签验证/保存解锁时用的sn token signature结果至0x2200-0x2400/rpmb检查lock-bit

那答案必然是用一个可持续的漏洞，直接改get_lock_status函数，已经没有必要调用set_lock_status写合法0x2000-0x2040数据了，反正其他验证必然过不去（这个改法下来，狭义上说并没解锁，但所有系统组件认为解了）

那要使得uboot/lk被改动，那漏洞只能在brom/chipram了（spl和fdl1都是chipram）

TomKing062, 2025-04-28:
要是能接受tethered boot，改完uboot/lk接着用38694没什么问题