TomKing062/CVE-2022-38694_unlock_bootloader #176: Locking the bootloader and bootloop when restoring from TWRP backup.

zoomver, 2025-04-17:
Hello.
I have a question for you.:
Smartphone ZTE Blade A31 (sc9863a, unisoc), I collect TWRP from sources on GitHub. When testing recovery from backup, the bootloader and bootloop are blocked. I performed a full backup, and also tried to exclude the following blocks from backup: misc, boot, uboot. After backup, I rebooted into the system, and also did a wipe before rebooting into the system. In all cases, the loader is blocked and a bootloop occurs.
If you have the opportunity, please tell me why this is happening.
Thank you very much if you take the time.
Good luck to you.

TomKing062, 2025-04-21:
unlock info is in miscdata

TomKing062, 2025-04-21:
slot_ab info in misc

zoomver, 2025-04-21:
Hello.
Thanks a lot for the hint.
Now I will try to restore backup without miscdata.
After conducting the experiments, I will definitely write to you what happened.
I hope it works out.
Thanks again.

zoomver, 2025-04-22:
Hello.
Thank you very much for the explanation and the hint.
I made a backup in TWRP without the "miscdata" file. Next, I made a recovery from backup.
Everything was successful.
The loader remained unblocked.
There was no bootloop.
Thank you very much. I am very grateful to you.
Good luck to you and all the best.

closed 2025-04-22 by zoomver