TomKing062/CVE-2022-38694_unlock_bootloader #177: Soft Bricked A7 after following instructions.

isaacjoel, 2025-04-18:
I ran:

.\adb reboot autodloader

Then installed libusb using Zadig and the SPD driver suggested in this repo's release notes. I also copied in the updated SPD dump tool to the ud710_hisense_a7/ directory. Then ran:

./unlock.bat 

Here is the output:

G:\Downloads\ud710_hisense_a7>setlocal enabledelayedexpansion

G:\Downloads\ud710_hisense_a7>if not exist "u-boot-spl-16k-sign.bin" (
spd_dump --wait 300 exec_addr 0x3f28 fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec r splloader r teecfg r trustos r sml e splloader e splloader_bak reset

 pause
 chsize splloader.bin
 if !errorlevel! EQU 0 (rename "splloader.bin" "u-boot-spl-16k-sign.bin" )
 chsize teecfg.bin
 chsize trustos.bin
 chsize sml.bin
 pause
)  else (
spd_dump --wait 300 exec_addr 0x3f28 fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec e splloader e splloader_bak reset

 pause
)
branch:stable, sha1:bb87079bd431462688f778b973bd7fa70fd80faa
hotplug unsupported on this platform
Waiting for dl_diag connection (300s)
libusb_control_transfer ok
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
current exec_addr is 0x3f28
SEND fdl1-dl.bin to 0x5500
SEND custom_exec_no_verify_3f28.bin to 0x3f28
EXEC FDL1
usb_recv failed : LIBUSB_ERROR_TIMEOUT
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
KEEP_CHARGE FDL1
SEND uboot-mod.bin to 0x9efffe00
FDL2: incompatible partition
EXEC FDL2
DISABLE_TRANSCODE
Reading Partition List
[=====                                   ] 12.5%
[==========                              ] 25.0%
[===============                         ] 37.5%
[====================                    ] 50.0%
[=========================               ] 62.5%
[==============================          ] 75.0%
[===================================     ] 87.5%
[========================================] 100.0%
Read Part Done: user_partition+0x0, target: 0x8000, read: 0x8000
  0                            splloader     256KB
  1                             miscdata       5MB
  2                                 misc       1MB
  3                            nr_fixnv1       8MB
  4                            nr_fixnv2       8MB
  5                               prodnv      10MB
  6                        nr_runtimenv1      10MB
  7                        nr_runtimenv2      10MB
  8                             recovery      40MB
  9                              trustos       6MB
 10                          trustos_bak       6MB
 11                                  sml       1MB
[... 309 lines cut ...]
[=============================           ] 72.7%
[==============================          ] 75.1%
[===============================         ] 77.5%
[================================        ] 80.1%
[=================================       ] 82.5%
[==================================      ] 85.1%
[===================================     ] 87.5%
[====================================    ] 90.1%
[=====================================   ] 92.6%
[======================================  ] 95.1%
[======================================= ] 97.6%
[========================================] 100.0%
Read Part Done: boot+0x0, target: 0x2300000, read: 0x2300000
File does not exist.
file size : 0x800
[========================================] 100.0%
Write Part Done: misc, target: 0x800, written: 0x800

G:\Downloads\ud710_hisense_a7>pause
Press any key to continue . . .
PS G:\Downloads\ud710_hisense_a7>

The phone didn't reboot and the output looks different to some other successful runs I've seen in other Issues. 

What am I doing wrong? Or at least, how can I restore my phone to it's original state and pretend this was all a bad dream?

TomKing062, 2025-04-21:
exec w splloader SPLFILE reset

find SPLFILE from xda or Hisense package

TomKing062, 2025-04-21:
fxxk `adb reboot autodloader`
never use it without backup

closed 2025-04-21 by TomKing062