TomKing062/CVE-2022-38694_unlock_bootloader #180: Which file to use for Tecno KL4 (UMS9230E)?

IsHacker003, 2025-04-20:
I tried "ums9230_universal_unlock.zip" but it didn't work. Result:

> ishacker@ISHR-PC:~/Downloads/ums9230_universal_unlock$ ./spd_dump --wait 300 exec_addr 0x65015f08 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0x9efffe00 exec r splloader r uboot e splloader e splloader_bak reset
branch:stable, sha1:bb87079bd431462688f778b973bd7fa70fd80faa
Waiting for dl_diag connection (300s)
libusb_control_transfer ok
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
current exec_addr is 0x65015f08
SEND fdl1-dl.bin to 0x65000800
SEND custom_exec_no_verify_65015f08.bin to 0x65015f08
EXEC FDL1
usb_recv failed : LIBUSB_ERROR_TIMEOUT
CHECK_BAUD FAIL
usb_recv failed : LIBUSB_ERROR_TIMEOUT
CHECK_BAUD FAIL
usb_recv failed : LIBUSB_ERROR_TIMEOUT
CHECK_BAUD FAIL
usb_send failed : LIBUSB_ERROR_TIMEOUT

TomKing062, 2025-04-21:
use `oem pull` to get spl and uboot(lk), thensend here
i don't know difference in UMS9230E yet

TomKing062, 2025-04-21:
oh btw Tecno has twice check during boot, i don't think you can unlock bootloader with CVE

TomKing062, 2025-04-21:
i just wonder if UMS9230E brom has changed (from UMS9230)

IsHacker003, 2025-04-21:
> use `oem pull` to get spl and uboot(lk), thensend here i don't know difference in UMS9230E yet

IsHacker003, 2025-04-21:
> use `oem pull` to get spl and uboot(lk), thensend here i don't know difference in UMS9230E yet
i pulled uboot_a and uboot_b (the phone has A/B partitions) but i get this for spl:

> ishacker@ISHR-PC:~ /python-fastboot-oem-pull$ python3 main.py oem pull spl_a
FastbootMessage(message=b'upload partition fail.', header=b'FAIL')
Traceback (most recent call last):
  File "/home/ishacker/python-fastboot-oem-pull/main.py", line 15, in <module>
    print(_protocol.HandleSimpleResponses(5000, DEFAULT_MESSAGE_CALLBACK))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ishacker/python-fastboot-oem-pull/adb/fastboot.py", line 88, in HandleSimpleResponses
    return self._AcceptResponses(info_cb, timeout_ms=timeout_ms)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ishacker/python-fastboot-oem-pull/adb/fastboot.py", line 115, in _AcceptResponses
    raise FastbootRemoteFailure('FAIL: %s', remaining)
adb.fastboot.FastbootRemoteFailure: ("FAIL: b'upload partition fail.'", b'upload partition fail.')
ishacker@ISHR-PC:~/python-fastboot-oem-pull$ python3 main.py oem pull spl_b
FastbootMessage(message=b'upload partition fail.', header=b'FAIL')
Traceback (most recent call last):
  File "/home/ishacker/python-fastboot-oem-pull/main.py", line 15, in <module>
    print(_protocol.HandleSimpleResponses(5000, DEFAULT_MESSAGE_CALLBACK))
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ishacker/python-fastboot-oem-pull/adb/fastboot.py", line 88, in HandleSimpleResponses
    return self._AcceptResponses(info_cb, timeout_ms=timeout_ms)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ishacker/python-fastboot-oem-pull/adb/fastboot.py", line 115, in _AcceptResponses
    raise FastbootRemoteFailure('FAIL: %s', remaining)
adb.fastboot.FastbootRemoteFailure: ("FAIL: b'upload partition fail.'", b'upload partition fail.')

[uboot.zip](https://github.com/user-attachments/files/19832678/uboot.zip)

TomKing062, 2025-04-21:
"splloader"

IsHacker003, 2025-04-21:
> "splloader"

ok, here is the file.

[splloader.zip](https://github.com/user-attachments/files/19832739/splloader.zip)

TomKing062, 2025-04-23:
not surprising
[image]

IsHacker003, 2025-05-11:
> not surprising [image]

i found a modified spd_dump.exe and custom fdl1-dl.bin and fdl2-dl.bin on 4PDA, and my bootloader is unlocked! i attached the zip file. can you add it in the releases?

[ums9230e_Tecno_KL4.zip](https://github.com/user-attachments/files/20147625/ums9230e_Tecno_KL4.zip)

TomKing062, 2025-05-11:
huh, so it's dram initialization issue