TomKing062/CVE-2022-38694_unlock_bootloader #183: 关于unisoc开启KVM的可能性

hicode002, 2025-04-26:
不清楚unisoc各个固件运行时候所处的异常等级，另外缺乏unisoc启动流程的资料，希望wiki可以更新一下或者给个链接
然后既然拿到了bootrom的权限，那么EL3下面可以控制hyp虚拟机管理程序吧，这样就能以EL2加载内核（然后编译KVM),或者直接以EL3加载内核，这样或许可以使用KVM虚拟化。不知道可行吗？

dann2333, 2025-05-02:
那你可能需要爆改下面一大堆组件，同样是没有任何资料的。
unisoc大致启动流程的图片csdn有一个

TomKing062, 2025-05-11:
讲启动流程的文档和代码都没有显式的表明EL等级
大致上就是
brom-fdl1-fdl2(uboot/lk) 下载模式是不带secure world
brom-spl-sml-uboot-(kernel-android) 这个是开secure隔离的
sml用于初始化trustos（及附加teecfg）
sysdump这种特殊的模式流程是brom-spl-uboot（然后uboot直接和pc通信数据）

通过spd_dump(+IDA)也能强行搞出brom-fdl1_dl-sml-fdl2(uboot/lk) 的下载模式
以及brom-fdl1_boot-sml-fdl2(uboot/lk)的启动模式

如果不加载sml（tos teecfg），在uboot内无法调用tee_common_call系列函数

sml实际上跟mtk的ATF机制挺类似的？

hicode002, 2025-06-06:
> 讲启动流程的文档和代码都没有显式的表明EL等级 大致上就是 brom-fdl1-fdl2(uboot/lk) 下载模式是不带secure world brom-spl-sml-uboot-(kernel-android) 这个是开secure隔离的 sml用于初始化trustos（及附加teecfg） [...]

所以说正常启动时候uboot是el3还是el2，还能不能读写spl的内存?

TomKing062, 2025-06-06:
实话：不知道（等我哪天编译一套spl+fdl1+uboot直接打印出来（1周-1月都有可能
brom-spl-sml-uboot(启动到bootloader模式)，能读到spl，读不到brom
brom-spl/fdl1启动完成，还能读到brom
brom-spl/fdl1-sml启动完成，不能读到brom

spl/fdl1内存是被专门保留在ram里的，是写入spl到分区时对spl进行校验用的

hicode002, 2025-06-06:
> 所以说正常启动时候uboot是el3还是el2，还能不能读写spl的内存?

嗯嗯现在我想问跳转到uboot加载地址以后也许还能读取spl的内存，但是还能不能写入spl的内存然后再切回spl重新加载分区（绕过安全启动情况下）？或者能不能在跳转到uboot加载地址后写入uboot自己区域的内存？也就是说跳转到uboot之后有没有内存写保护？

TomKing062, 2025-06-09:
虽然你已经知道了，我写出来给其他人看下（
跳转到uboot地址的这一时刻，uboot区域没有写保护，至于uboot自己执行到后期有没有给自己的代码段加保护，不清楚（

TomKing062, 2025-07-28:
[image]

[image]

hicode002, 2026-01-11:
> 发件时间：2026年1月11日 21:52
>
>
> 默认el2，不排除厂商配置
>  参考
>  https://github.com/Veynamer/bsp_bootloader_u-boot15_sharkl5pro/blob/eba43fa3542553a7d493edd9c1a47e423b9c5b28/arch/arm/cpu/armv8/start.S#L191
>
> 展讯BSP内核对kvm支持很少，UART这玩意似乎与tx接地冲突，至少我在真机上接上就进u2s，启动后再接也无输出，可能量产禁用了调试UART，uboot_log写日志似乎是在最后阶段，uboot代码死机会无输出写入

Thanks!I have tried enabling kvm and got success.

最美的云
***@***.***

        

         原始邮件