TomKing062/CVE-2022-38694_unlock_bootloader #185: Ponsel saya tidak bisa di hidupkan setelah menjalankan unlock auto patch

PrinceMadds, 2025-05-09:
Hello I am using a realme c33 phone and I have installed the spdr driver and I have run the unlock auto patch, when I plug the usb into my phone at that time it could not connect perfectly, and I unplugged the usb from my phone and tried with another cable, and it worked after the process was running and finished my phone could not be turned on and charged also could not what is the solution please help me

TomKing062, 2025-05-11:
find realme c33 pac firmware or ask after sale service

i don't understand “running and finished” , since the batch itself should backup splloader(u-boot-spl-16k) and uboot(u-boot)

PrinceMadds, 2025-05-11:
> temukan firmware realme c33 pac atau tanyakan layanan purna jual
> 
> saya tidak mengerti "berjalan dan selesai", karena batch itu sendiri harus mencadangkan splloader(u-boot-spl-16k) dan uboot(u-boot

> find realme c33 pac firmware or ask after sale service
> 
> i don't understand “running and finished” , since the batch itself should backup splloader(u-boot-spl-16k) and uboot(u-boot)

do i have to re-flash the firmware?

BocahBocah, 2025-06-16:
> do i have to re-flash the firmware?

https://sfile.mobi/8mAw6MADdYU

https://sfile.mobi/9FQwQaF43KW
https://sfile.mobi/6sov4eGHD2k