TomKing062/CVE-2022-38694_unlock_bootloader #191: 脚本逻辑性问题

DDZH-DEV, 2025-06-01:
近期在使用 ums9620_ZTE_universal过程中，逐步分析
`
if not exist "u-boot-spl-16k-sign.bin" (
	spd_dump --wait 300 exec_addr 0x65012f48 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0xb4fffe00 exec r splloader r uboot e splloader e splloader_bak reset
	@echo "(This is a notice, not an error.) Don't continue if you see find port failed, just close and re-run this batch"
	pause
	gen_spl-unlock splloader.bin
	if !errorlevel! equ 0 (
		rename "splloader.bin" "u-boot-spl-16k-sign.bin"
		chsize uboot.bin
		rename uboot.bin uboot_bak.bin
	)
	pause
) else (
	spd_dump --wait 300 exec_addr 0x65012f48 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0xb4fffe00 exec e splloader e splloader_bak reset
	@echo "(This is a notice, not an error.) Don't continue if you see find port failed, just close and re-run this batch"
	pause
)
`

gen_spl-unlock splloader.bin  好像会直接生成 spl-unlock.bin ，但是不会rename 重命名u-boot-spl-16k-sign.bin，但是由于前面执行了 r splloader r uboot e splloader e splloader_bak  ，所以很多人都会造成重复执行，导致 splloader uboot 都会丢失，这里逻辑要改一下,避免更多的人备份丢失

还有脚本好像解不了 7551N   

[bin.zip](https://github.com/user-attachments/files/20538242/bin.zip)

DDZH-DEV, 2025-06-01:
求大佬指点

DDZH-DEV, 2025-06-01:
还有不解锁，强制性刷修补boot 签名有点问题 

[image]

[image]

`python.exe avbtool add_hash_footer --image AIK/image-new.img --partition_name boot --partition_size 67108864 --key rsa4096_boot.pem --algorithm SHA256_RSA4096  --prop com.android.build.boot.os_version:13   --prop com.android.build.boot.fingerprint:ZTE/CN_P720S15/P720S15:13/TP1A.220624.014/20241126.125516:user/release-keys  --rollback_index 0 --salt 8aa09babd7345a925546dba354dbd787a71b67c4de45801e5b13a891f2c7064d `  

--rollback_index 0还是1 无效  ,Prop: com.android.build.boot.security_patch -> '2024-11-05'
    Prop: com.android.build.system.os_version -> '13'
    Prop: com.android.build.system.security_patch -> '2024-11-05'
    Prop: com.android.build.vendor.os_version -> '13'
    Prop: com.android.build.vendor.security_patch -> '2024-11-05'  附加参数 加上也无效，直接提示 no valid os ,
如果不解锁单独关闭vb 校验无法重启

[image]

[vb-boot.zip](https://github.com/user-attachments/files/20539042/vb-boot.zip)

DDZH-DEV, 2025-06-02:
我切到win10下，可以完整跑一遍流程，跑完后没有解锁

TomKing062, 2025-06-02:
没觉得有问题，确实解了，也确实被回锁了
直接一点，fxxk ZTE

closed 2025-06-02 by TomKing062

hicode002, 2025-06-05:
> 不复原备份你怎么开机，cve是单次的又不是持久性攻击

不对啊不应该擦除misc分区吧，毕竟解锁状态储存在misc分区 

> 近期在使用 ums9620_ZTE_universal过程中，逐步分析
> 
> ```
> if not exist "u-boot-spl-16k-sign.bin" (
> 	spd_dump --wait 300 exec_addr 0x65012f48 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0xb4fffe00 exec r splloader r uboot e splloader e splloader_bak reset
> 	@echo "(This is a notice, not an error.) Don't continue if you see find port failed, just close and re-run this batch"
> 	pause
> 	gen_spl-unlock splloader.bin
> 	if !errorlevel! equ 0 (
> 		rename "splloader.bin" "u-boot-spl-16k-sign.bin"
> 		chsize uboot.bin
> 		rename uboot.bin uboot_bak.bin
> 	)
> 	pause
> ) else (
> 	spd_dump --wait 300 exec_addr 0x65012f48 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0xb4fffe00 exec e splloader e splloader_bak reset
> 	@echo "(This is a notice, not an error.) Don't continue if you see find port failed, just close and re-run this batch"
> 	pause
> )
> ```
> 
> gen_spl-unlock splloader.bin 好像会直接生成 spl-unlock.bin ，但是不会rename 重命名u-boot-spl-16k-sign.bin，但是由于前面执行了 r splloader r uboot e splloader e splloader_bak ，所以很多人都会造成重复执行，导致 splloader uboot 都会丢失，这里逻辑要改一下,避免更多的人备份丢失
> 
> 我执行完 ` spd_dump exec_addr 0x65012f48 fdl spl-unlock.bin 0x65000800 ::check unlock (if get 64 zeros, locked; if 32 string + 16 hash + 16 hash, unlocked)
> 
> spd_dump exec_addr 0x65012f48 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0xb4fffe00 exec verbose 2 read_part miscdata 8192 64 m.bin reset
> 
> ` 得到的是
> 
> [image] 看上去是解锁了，但是最后一句 ` spd_dump exec_addr 0x65012f48 fdl fdl1-dl.bin 0x65000800 fdl fdl2-dl.bin 0xb4fffe00 exec r boot w splloader u-boot-spl-16k-sign.bin w uboot uboot_bak.bin w misc misc-wipe.bin reset
> 
> ` 执行后又会复原，没有解锁标志，而不刷，不恢复又不能开机，看不明白了
> 
> [bin.zip](https://github.com/user-attachments/files/20538242/bin.zip)

There exists two partitions misc and miscdata.Unlock state stores in miscdata not misc

madobet, 2025-06-08:
我想问问 splloader uboot 丢失了怎么办，e splloader e splloader_bak 这俩玩意执行完全都给扬了吧？感谢脚本，已经砖掉两台手机几百块钱搭进去了

TomKing062, 2025-06-08:
哦 zte脚本确实加了adb reboot autodloader
那是因为我知道中兴的9620运行这句不擦除spl
非中兴的你看看通解dram1和dram2
那俩没加adb reboot autodloader

madobet, 2025-06-08:
> 哦 zte脚本确实加了adb reboot autodloader 那是因为我知道中兴的9620运行这句不擦除spl 非中兴的你看看通解dram1和dram2 那俩没加adb reboot autodloader

回的好快，不是zte的，是 T7510，脚本类似反正。

另外弄了台设备，死马当活马医把 splloader/fdl1 和 uboot/fdl2 重新 dump 做了下，意外救好了。也不好说是不是 usb hub 或 usb线本身的问题。

而且有个很奇怪的事，救砖以后 r splloader 命令下来的 splloader.bin 通不过 chsize 了… 原来是 59K 的 u-boot-spl-16k-sign.bin，同样的命令现在 dump 下来的 splloader.bin 是 256KB 的了。