TomKing062/CVE-2022-38694_unlock_bootloader #193: ITEL S23 (S665L) not turning on after UBL | Stuck in SPRD U25 Diag...

Kumardeshpande, 2025-06-07:
=======================================
        Unlock Bootloader
 =======================================
Perhatian Penggunaan :
- Matikan Hp anda
- Tekan volume bawah
- sambung Hp anda
- selama proses jangan lepas volume bawah
=========================================
jika sudah siap klik apa saja untuk melanjutkan
Press any key to continue . . .
=======================================
      MENCADANGKAN Dan MENGHAPUS SPLLOADER
perhatian : setelah ini hp tidak akan bisa hidup
=======================================
=======================================
    MENGHAPUS SPLLOADER (karena splloader backup sudah ada
perhatian : setelah ini hp tidak akan bisa hidup
=======================================
branch:main, sha1:7d5550b97b8a7f74ccb35139665af94a30818ce3
Successfully connected to port: 12
PERIKSA BAUD MODE BOOTROM
BSL_REP_VER: "SPRD3\0"
CMD TERSAMBUNG MODE BOOTROM
exec_addr Sekarang adalah 0x65015f08
SEND fdl1-dl.bin to 0x65000800
SEND custom_exec_no_verify_65015f08.bin to 0x65015f08
MENJALANKAN FDL1
PERIKSA BAUD MODE FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD TETSAMBUNG MODE FDL1
MENGUBAH BAUD MODE FDL1 KE 921600
KEEP_CHARGE FDL1
SEND fdl2-dl.bin to 0x9efffe00
FDL2: incompatible partition
MENJALANKAN FDL2
DISABLE_TRANSCODE
Reading Partition List
[=====                                   ] 12.5%
[==========                              ] 25.0%
[===============                         ] 37.5%
[====================                    ] 50.0%
[=========================               ] 62.5%
[==============================          ] 75.0%
[===================================     ] 87.5%
[========================================] 100.0%
Cadangan Partisi Berhasil: user_partition+0x0, target: 0x8000, read: 0x8000
  0                            splloader     256KB
  1                               prodnv      10MB
  2                             miscdata       1MB
  3                                 misc       1MB
  4                            trustos_a       6MB
  5                            trustos_b       6MB
  6                                sml_a       1MB
  7                                sml_b       1MB
  8                              uboot_a       3MB
  9                              uboot_b       3MB
 10                            uboot_log       4MB
 11                                 logo       8MB
 12                            fbootlogo      12MB
 13                           l_fixnv1_a       2MB
 14                           l_fixnv2_a       2MB
 15                           l_fixnv1_b       2MB
 16                           l_fixnv2_b       2MB
 17                         l_runtimenv1       2MB
 18                         l_runtimenv2       2MB
 19                          gnssmodem_a       1MB
 20                          gnssmodem_b       1MB
 21                           wcnmodem_a      10MB
 22                           wcnmodem_b      10MB
 23                              persist       2MB
 24                            l_modem_a      25MB
 25                            l_modem_b      25MB
 26                          l_deltanv_a       1MB
 27                          l_deltanv_b       1MB
 28                             l_gdsp_a      10MB
 29                             l_gdsp_b      10MB
 30                             l_ldsp_a      20MB
 31                             l_ldsp_b      20MB
 32                            l_agdsp_a       6MB
 33                            l_agdsp_b       6MB
 34                             pm_sys_a       1MB
 35                             pm_sys_b       1MB
 36                             teecfg_a       1MB
 37                             teecfg_b       1MB
 38                         hypervsior_a      10MB
 39                         hypervsior_b      10MB
 40                               boot_a      64MB
 41                               boot_b      64MB
 42                        vendor_boot_a     100MB
 43                        vendor_boot_b     100MB
 44                          init_boot_a       8MB
 45                          init_boot_b       8MB
 46                                dtb_a       8MB
 47                                dtb_b       8MB
 48                               dtbo_a       8MB
 49                               dtbo_b       8MB
 50                                super    5600MB
 51                                cache     100MB
 52                             vbmeta_a       1MB
 53                             vbmeta_b       1MB
 54                             metadata      16MB
 55                            sysdumpdb      20MB
 56                      vbmeta_system_a       1MB
 57                      vbmeta_system_b       1MB
 58                      vbmeta_vendor_a       1MB
 59                      vbmeta_vendor_b       1MB
 60                  vbmeta_system_ext_a       1MB
 61                  vbmeta_system_ext_b       1MB
 62                     vbmeta_product_a       1MB
 63                     vbmeta_product_b       1MB
 64                         vbmeta_odm_a       1MB
 65                         vbmeta_odm_b       1MB
 66                         avbmeta_rs_a       1MB
 67                         avbmeta_rs_b       1MB
 68                         common_rs1_a       8MB
 69                         common_rs1_b       8MB
 70                         common_rs2_a      16MB
 71                         common_rs2_b      16MB
 72                               tranfs      50MB
 73                             userdata  115727MB
Tabel GPT standar disimpan pgpt.bin
Melewati Menyimpan Paket Daftar Partisi SPRD
Daftar partisi disimpan ke partition_1749323635.xml
Jumlah Seluruh partisi: 73
Tipe Memori device ini UFS
ENABLE_WRITE_RAW_DATA
perangkat menggunakan a
Erase Part Done: splloader
Erase Part Done: splloader_bak

Waiting for 0 seconds, press CTRL+C to quit ...
=======================================
=======================================
Membuat File Spl Unlock
....
....
The file is not sprd trusted firmware

Waiting for 0 seconds, press CTRL+C to quit ...
....
....
=======================================
=======================================
        MEMASANG UBOOT MODIFIKASI
=======================================
branch:main, sha1:7d5550b97b8a7f74ccb35139665af94a30818ce3
Successfully connected to port: 12
PERIKSA BAUD MODE BOOTROM
BSL_REP_VER: "SPRD3\0"
CMD TERSAMBUNG MODE BOOTROM
exec_addr Sekarang adalah 0x65015f08
SEND fdl1-dl.bin to 0x65000800
SEND custom_exec_no_verify_65015f08.bin to 0x65015f08
MENJALANKAN FDL1
PERIKSA BAUD MODE FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD TETSAMBUNG MODE FDL1
MENGUBAH BAUD MODE FDL1 KE 921600
KEEP_CHARGE FDL1
SEND fdl2-dl.bin to 0x9efffe00
FDL2: incompatible partition
MENJALANKAN FDL2
DISABLE_TRANSCODE
Reading Partition List
Error, Batas waktu Tercapai
Press any key to continue . . .
Press any key to continue . . .
=======================================
=======================================
      MENGIRIM UNLOCKER BOOTLOADER
perhatian : jika terdapat kata perangkat dilepas berarti berhasil
=======================================
branch:main, sha1:7d5550b97b8a7f74ccb35139665af94a30818ce3
Successfully connected to port: 12
pengiriman_usb gagal (0 / 1)
Press any key to continue . . .
Press any key to continue . . .
=======================================
=======================================
      MEMBACA STATUS BOOTLOADER
recv 64 zero = locked
recv 32 string + 16 hash + 16 hash (acak acakan) = unlocked
=======================================
branch:main, sha1:7d5550b97b8a7f74ccb35139665af94a30818ce3
Successfully connected to port: 12
pengiriman_usb gagal (0 / 1)
Press any key to continue . . .

Waiting for  0 seconds, press CTRL+C to quit ...
=======================================
=======================================
  \\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\

1. Using this utility devices bootloader unlocked successfully but device not powring on its stuck in SPD Diga Port (Download Mode).
2. i was triyed below commans as guide 
"copy u-boot-spl-16k-sign.bin to spl.bin only execute last command"
spd_dump fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec erase_part uboot_log write_part splloader spl.bin timeout 100000 write_part userdata userdata.bin reset
But not success so please guaide how fix this problem???????

TomKing062, 2025-06-09:
find a valid splloader backup(or full firmware)
internet or aftersale service or some paid toolbox

in normal case, “poweroff, then vol_down_key” won't erase splloader

Massatriof16, 2025-06-16:
BRO IS IT MY BAT MENU FOR EXECUTE SPD_DUMP???

closed 2025-08-27 by TomKing062