TomKing062/CVE-2022-38694_unlock_bootloader #198: Alldocube iPlay 60 fails to unlock

tymercer, 2025-06-30:
Using the 9230 universal as well as the iPlay 50 mini, fails with the following log entries.
Attached screenshots of the cmd output

[2025-06-30 02:38:16:911] Read: Call WaitCommEvent()
[2025-06-30 02:38:17:878] Read: Call WaitForSingleObject() and not return WAIT_TIMEOUT, [Return:0x00000000][EventMask:0x00000000].
[2025-06-30 02:38:17:879] Read: Call ClearCommError() fail, [ErrorCode: 0x000003E3].
[2025-06-30 02:38:17:879] Write: ClearCommError
[2025-06-30 02:38:17:879] Write: Call ClearCommError() fail, [ErrorCode: 0x00000016].
[2025-06-30 02:38:17:879] Write: Call WriteFile() fail, [ErrorCode: 0x00000016].
[2025-06-30 02:41:45:252] Write: Call GetOverlappedResult() fail, [ErrorCode: 0x000001B1].

[image]

[image]

To note:
This device fails the following as well
fastboot flashing unlock
fastboot oem unlocking, etc.

Let me know if there are any specific files/etc you need to help with this.
Thanks

TomKing062, 2025-07-18:
T1102T is not unisoc? google said it's MTK

TomKing062, 2025-07-28:
the massage seems device is not dl_diag
dl_diag need key pressed
the u2s shows up during boot (without key) is boot_diag

if fdl1/fdl2 fails ,you need to extract from Alldocube provided PAC firmware

tymercer, 2025-07-28:
I'm not sure what that means, can you give me some guidance on what needs to be done?

closed 2025-08-27 by TomKing062