TomKing062/CVE-2022-38694_unlock_bootloader #199: a5procc 一直卡在下载模式

6lj6, 2025-07-16:
我亲爱的技术高超、爱心澎湃的 汤姆王：

* 一开始尝试 ubuntu xia，之前反复组合键，然后一行行运行 unlock.bat 里的命令
* 然后就卡在某个充电图像不动了

尝试安装 Windows 10，一键 unlock.bat ，数据在跑动，但是还是无法开机。

看了历史 issue，应该是卡在下载模式了，但好像当时那个人也没做什么就偶然好了。

目前我护眼的 a5procc 一直显示充电54%，无奈用起了 OLED 屏幕的手机，眼睛难受，万望解救。

6lj6, 2025-07-17:
鉴于 本仓库wiki 为了通用性而省略了部分细节，我将在本 issue 记录 有关 a5procc 型号的使用本仓库的具体操作。

1. 手机的下载模式，或升级模式。
    * 是在关机情况下后，快捷键是”先同时按住“音量+”和“音量-”两个侧键不放，再按下开机键“。
    * 而非笼统的 “POWER + VOL_UP + VOL_DOWN ”，事实上这三个键长按会导致电脑有时能连接、有时检测不到。
    * 故 wiki 的最佳描述是”VOL_UP + VOL_DOWN + POWER"
    * 备注：该操作来自官方 https://fans.hisense.com/forum.php?mod=viewthread&tid=172400&highlight=a5procc
2. 如果需要 rom，官网的第一版 链接：https://pan.baidu.com/s/1mffUWCDb_wGxv0BZVQ5IRg 提取码：3ir6
    * 目前用本仓库的 wiki 引导已导致手机卡屏，可能这个方法能回退，故下载尝试中 。
    * 备注：该链接来自官方 https://fans.hisense.com/forum.php?mod=viewthread&tid=171332&highlight=a5procc
3. 网上有其他 root 方式，或者本仓库的老版本介绍，造成错综复杂的操作、命令的修改，为了保持与本仓库一致，严格按照本仓库的最新 wiki 进行操作（安全、可靠）
4. 历史有关 a5procc 的 issue
    * 很多细小改动不知是否已合并最新的命令
5. 广泛测试。
    * 每个步骤的成功或失败的标志不清晰。

目前bat 运行的时候很多地方都显示 failed，如果跟目标无关，这些 failed 可以不显示。（或者在教程中写明这些信息可以不用关心）

但目前出问题了后，这些消息就不知道是哪个可能有作用了。

```
C:\Users\anon\Desktop\a5procc\tomking_spd\ums512_hisense_a5procc>spd_dump --wait 300 exec_addr 0x3ee8 fdl fdl1-boot.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 fdl teecfg.bin 0x9401fe00 fdl trustos.bin 0x9403fe00 fdl sml.bin 0x93fffe00 exec
branch:main, sha1:fa0becf5e3f026b3b99103c65de6eb9a8348b27c
Successfully connected to port: 5
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
current exec_addr is 0x3ee8
SEND fdl1-boot.bin to 0x5500
SEND custom_exec_no_verify_3ee8.bin to 0x3ee8
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
CHANGE_BAUD FDL1 to 921600
KEEP_CHARGE FDL1
SEND uboot-mod.bin to 0x9efffe00
SEND teecfg.bin to 0x9401fe00
SEND trustos.bin to 0x9403fe00
SEND sml.bin to 0x93fffe00
timeout reached
```

6lj6, 2025-07-17:
因为 hovatek 的方案也是基于本仓库，相当于本仓库的 a5procc 型号的详细版本，现已 root 成功，注意点如下：
* root https://www.hovatek.com/forum/thread-46000.html
* 注意下载模式快捷键启动后毫无反应（但实际已经启动成功，按住音量加减电源键后七秒左右赶紧连线）
* 注意其中先改为 image-new.img ，sign 命令后，再改回原 patch 好的包名进行 send
* 注意如果上面没操作，会导致卡第一屏，官方有给出方案 https://fans.hisense.com/thread-207030-1-1.html ，即

> 如果卡第一屏了，可以用按键的方式重新进入 bootloader 模式，方法为保证 USB 线连接的情况下长按电源键+上音量键。

目前已经 root 成功，感谢以上各位的分享。

[\[Hovatek\] Hisense_a5proccc BL unlock.zip](https://github.com/user-attachments/files/21299137/Hovatek.Hisense_a5proccc.BL.unlock.zip)

[\[Hovatek\] Hisense A5 Pro CC Root.zip](https://github.com/user-attachments/files/21299126/Hovatek.Hisense.A5.Pro.CC.Root.zip)

wei-coding, 2025-07-21:
我在一開始的步驟忘記進入開發者選項開啟unlock bootloader的選項，現在一樣卡在download mode裡，這樣還有救嗎?

nullscode, 2025-08-13:
> 我在一開始的步驟忘記進入開發者選項開啟unlock bootloader的選項，現在一樣卡在download mode裡，這樣還有救嗎?

有救的,找到对应的splloader.bin文件刷进去就能复活 这个文件在官方刷机包里面有,不过名字改成了u-boot-spl-16k.bin

closed 2025-08-13 by TomKing062