TomKing062/CVE-2022-38694_unlock_bootloader #210: a7cc刷成砖了,一直卡在sprd u2s diag 无论怎么重启都这样,无法回到fastboot了.bat

nullscode, 2025-08-13:
C:\Users\h\Documents\文件夹\a7cc解锁>setlocal enabledelayedexpansion

C:\Users\h\Documents\文件夹\a7cc解锁>if not exist "u-boot-spl-16k-sign.bin" (
spd_dump --wait 300 exec_addr 0x3f28 fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec r splloader r teecfg r trustos r sml e splloader e splloader_bak reset

 pause
 chsize splloader.bin
 if !errorlevel! EQU 0 (rename "splloader.bin" "u-boot-spl-16k-sign.bin" )
 chsize teecfg.bin
 chsize trustos.bin
 chsize sml.bin
 pause
)  else (
spd_dump --wait 300 exec_addr 0x3f28 fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec e splloader e splloader_bak reset

 pause
)
branch:main, sha1:fa0becf5e3f026b3b99103c65de6eb9a8348b27c
Waiting for dl_diag connection (300s)
Successfully connected to port: 10
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
current exec_addr is 0x3f28
SEND fdl1-dl.bin to 0x5500
SEND custom_exec_no_verify_3f28.bin to 0x3f28
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
CHANGE_BAUD FDL1 to 921600
KEEP_CHARGE FDL1
SEND uboot-mod.bin to 0x9efffe00
FDL2: incompatible partition
EXEC FDL2
DISABLE_TRANSCODE
Reading Partition List
[=====                                   ] 12.5%
[==========                              ] 25.0%
[===============                         ] 37.5%
[====================                    ] 50.0%
[=========================               ] 62.5%
[==============================          ] 75.0%
[===================================     ] 87.5%
[========================================] 100.0%
Read Part Done: user_partition+0x0, target: 0x8000, read: 0x8000
  0                            splloader     256KB
  1                             miscdata       5MB
  2                                 misc       1MB
  3                            nr_fixnv1       8MB
  4                            nr_fixnv2       8MB
  5                               prodnv      10MB
  6                        nr_runtimenv1      10MB
  7                        nr_runtimenv2      10MB
  8                             recovery      40MB
  9                              trustos       6MB
 10                          trustos_bak       6MB
 11                                  sml       1MB
 12                              sml_bak       1MB
 13                                uboot       1MB
 14                            uboot_bak       1MB
 15                            uboot_log       4MB
 16                                 logo       7MB
 17                            fbootlogo       7MB
 18                              l_pmsys       1MB
 19                              l_agdsp       6MB
 20                            gnssmodem       1MB
 21                             wcnmodem      10MB
 22                              persist       2MB
[... 404 lines cut ...]
[=======================                 ] 57.6%
[========================                ] 60.0%
[=========================               ] 62.6%
[==========================              ] 65.0%
[===========================             ] 67.6%
[============================            ] 70.1%
[=============================           ] 72.7%
[==============================          ] 75.1%
[===============================         ] 77.5%
[================================        ] 80.1%
[=================================       ] 82.5%
[==================================      ] 85.1%
[===================================     ] 87.5%
[====================================    ] 90.1%
[=====================================   ] 92.6%
[======================================  ] 95.1%
[======================================= ] 97.6%
[========================================] 100.0%
Read Part Done: boot+0x0, target: 0x2300000, read: 0x2300000
File does not exist.
file size : 0x800
[========================================] 100.0%
Write Part Done: misc, target: 0x800, written: 0x800

C:\Users\h\Documents\文件夹\a7cc解锁>pause

nullscode, 2025-08-13:
找了很多资料 最后发现把这个文件刷到splloader 能救活 分享一下,希望能帮上遇到同样问题的人

[u-boot-spl-16k.zip](https://github.com/user-attachments/files/21751944/u-boot-spl-16k.zip)

closed 2025-08-13 by nullscode

sunjiahao5340, 2025-11-27:
> 找了很多资料 最后发现把这个文件刷到splloader 能救活 分享一下,希望能帮上遇到同样问题的人
> 
> [u-boot-spl-16k.zip](https://github.com/user-attachments/files/21751944/u-boot-spl-16k.zip)

我刷了这个固件，还是不能开机

snowfalcon11, 2026-02-08:
> [@snowfalcon11](https://github.com/snowfalcon11) 首先，请确保程序文件夹中不包含您自己的 u-boot-spl-16k.bin 文件。
> 
> 请尝试在命令行中，于程序文件夹内运行以下命令： `spd_dump fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec w splloader w splloader.bin w splloader u-boot-spl-16k-sign.bin w uboot uboot_bak.bin`
> 
> 如果这不起作用，请将 u-boot-spl-16k.zip 压缩包中的 u-boot-spl-16k.bin 文件移动到程序文件夹，然后运行另一个命令 `spd_dump fdl fdl1-dl.bin 0x5500 fdl uboot-mod.bin 0x9efffe00 exec w splloader u-boot-spl-16k-sign.bin`

非常感谢！ 
不过可能是我的文件不对，一直卡在这里
Successfully connected to port: 5
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
SEND fdl1-dl.bin to 0x5500
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FDL1
BSL_REP_VER: "Spreadtrum Boot Block version 1.1\0"
CMD_CONNECT FDL1
CHANGE_BAUD FDL1 to 921600
KEEP_CHARGE FDL1
SEND fdl1-boot.bin to 0x9efffe00
timeout reached

Shocked-Cat, 2026-02-08:
@snowfalcon11 
尝试使用不同的 USB 端口和数据线。您可能需要在另一台电脑上进行尝试。