TomKing062/CVE-2022-38694_unlock_bootloader #215: 3x CHECK_BAUD FAIL on Hisense Q5

SEM0001, 2025-08-22:
Hi, I've been trying to unlock the BL on my Hisense Q5, with haven't had much luck so far. This is the output I get every single time (3x CHECK_BAUD FAIL after EXEC FDL1). I've tried on both Windows 11 and Windows 7, using different USB cables and ports, and with both the official and libusb drivers.

`branch:main, sha1:fa0becf5e3f026b3b99103c65de6eb9a8348b27c
Waiting for dl_diag connection (300s)
Successfully connected to port: 5
CHECK_BAUD bootrom
BSL_REP_VER: "SPRD3\0"
CMD_CONNECT bootrom
current exec_addr is 0x3ee8
SEND fdl1-dl.bin to 0x5500
SEND custom_exec_no_verify_3ee8.bin to 0x3ee8
EXEC FDL1
CHECK_BAUD FAIL
CHECK_BAUD FAIL
CHECK_BAUD FAIL
usb_send failed (0 / 1)`

Some guides for the Hisense A5 Pro (same SoC) suggest changing the address in the first spd_dump command from `0x3ee8` to `0x3f48`, especially when entering download mode with buttons (Vol+ + Vol– + Power) instead of through adb. The thing is, those guides aren't very recent and the unlock package is different now, so I'm not sure if that's still safe to follow. Also, I don't think I can use `adb reboot autodloader` since I don't have an splloader dump to write back. What can be done here? Would appreciate any help!

TomKing062, 2025-08-22:
try delete 'exec_addr 0x3ee8'
there is some conflict between 38694(exec addr) and 38691(embedded in fdl1)

SEM0001, 2025-08-22:
Thank you very much! I removed the exec_addr from the first command and this time it worked without a hitch.

[image]