TomKing062/CVE-2022-38694_unlock_bootloader #221: Help unlocking the Oukitel G2

RigacciOrg, 2025-08-28:
Can someone please guide me in using this tool to unlock the bootloader of my **Oukitel G2** smartpone? I think it is based on the **Unisoc T606 ums9230**.

I downloaded the official pac ROM and unpacked it; into the unpack directory I find several files, e.g. **FileList.ini** contains some info like this:

```
[PAC_ums9230_6h10]
...
FDL=1@fdl1-sign.bin@0xEDF8@0x1CDD0
...
FDL2=1@lk-fdl2-sign.bin@0x1270D8@0x2BBC8
...
```

I'm a newbie with the Unisoc (I rooted only Mediatek devices), so the [unlock guide](https://github.com/TomKing062/CVE-2022-38694_unlock_bootloader/wiki/Unlock) is rather cryptic for me.

- I compiled the programs **chsize**, **gen_spl-unlock**, **gen_spl-unlock-legacy** and **spd_dump** on my Linux box.
- I copied the programs into an empty directory and now I have to put some **bin files** and create the **unlock_autopatch_9230.bat** file, but I can't figure out what files are required and how to write the batch file (I know the batch is for Windows, but I will execute the commands interactively in Linux).

Please help me with some hints or point me to the documentation that I have to read.

RigacciOrg, 2025-08-28:
I attach here two files extracted from the stock ROM, just in case they are usefull.

- [FileList.ini.txt](https://github.com/user-attachments/files/22022378/FileList.ini.txt)
- [PackInfo.csv](https://github.com/user-attachments/files/22022373/PackInfo.csv)

TomKing062, 2025-09-17:
gen_fdl1-dl for fdl1-sign
fdl2-dl and fdl2-cboot need manual patch
gen_spl-unlock for u-boot-spl-16k-sign/spl dumped from device

closed 2025-10-02 by TomKing062