TomKing062/CVE-2022-38694_unlock_bootloader #222: a7cc的分区表不知道怎么回事就坏了

nullscode, 2025-08-29:
[image]
今天在操作a7cc的时候,不知道怎么回事,突然就连不上了,每次都提示这句分区表损坏,然后就没响应了
我没有进行任何跟分区表有关的操作,不知道怎么回事,突然就这样了.
有啥办法救砖不?

nullscode, 2025-08-29:
send (8):
7e 00 04 00 00 ff fb 7e          |~......~|
recv (264):
7e 00 96 01 00 02 00 00 00 01 00 00 00 06 00 00  |~...............|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  |................|
00 00 00 00 00 f5 69 7e          |......i~|
FDL2: incompatible partition
EXEC FDL2
send (8):
7e 00 0d 00 00 ff f2 7e          |~......~|
send (8):
7e 00 21 00 00 ff de 7e          |~.!....~|
usb_send failed (0 / 8)

nullscode, 2025-08-29:
经过一些分析,发现那个incompatible partition提示就是个常规提示,分区没问题的情况下也会出现,现在问题变成: 发送完uboot-mod.bin之后就卡死了,没回应.

nullscode, 2025-08-29:
从中午分析到现在,目前判断是uboot-mod.bin里面可能有什么bug,在flash处于某些状态的情况下,就会卡死. 希望大佬能帮忙看看

nullscode, 2025-09-02:
我给弄了个pr
加了个参数 可以跳过0d指令

TomKing062, 2025-09-17:
fixed in spd_dump_it