TomKing062/CVE-2022-38694_unlock_bootloader #235: realme c53 ums9230_hulk bootloader unlocked with CVE-2022-38694_unlock_bootloader

cooked71, 2025-09-26:
Hi is it possible to flash and boot a custom rom  with this unlocked bootloader ? it seems that is unable to boot basically to this 00030235] [c0] avb_slot_verify() load_and_verify_vbmeta return ERROR while not allow verification error
[00030235] [c0] avb_slot_verify() avb slot verify FAILED!
[00030235] [c0] avb_slot_verify() fail label:slot_data is not NULL, free it
[00030235] [c0] avb_slot_verify() fail label:additional_cmdline_subst is not NULL, free it
[00030235] [c0] >>>>>>>>>>>>>>>>  avb_slot_verify result is 5.
[00030235] [c0] >>>>>>>>>>>>>>>>  avb_slot_verify result is ERROR_PUBLIC_KEY_REJECTED.
[00030235] [c0] slot_data[0] is 0x0.
[00030235] [c0] slot_data[1] is 0x0.
[00030235] [c0] avb_check_image() ret is 5.
[00030235] [c0] resolved_hashtree_error_mode = 0.
[00030235] [c0] keyword: = com.android.build.boot.security_patch key_size = 37
[00030235] [c0] WARNNING: LOCK FLAG IS : UNLOCK, SKIP VERIFY!!!
[00030235] [c0] [sprdlk][lcd_printf] enter lcd_printf
[00030240] [c0] INFO: your device is corrupt, It can't be trusted and may not work properly!
[00030240] [c0] [sprdlk][lcd_printf] enter lcd_printf
[00030246] [c0] INFO: first scene of dm-verity corrupt, start minidump...  is an issue on my build or is a partial unlock ? thanks

TomKing062, 2025-10-02:
disable-verity（verity 0） is vbmeta flag 1
disable-verification is flag 2
flag 3 is set both

unisoc only allow flag 1

closed 2025-10-02 by TomKing062

TomKing062, 2025-10-06:
boot/vendor_boot are controllable with BL-unlock
super is controllable with disable-verity

what other permission needed

cooked71, 2025-10-06:
i'm unsure if my  lineage 20 rom is malformed  (missing hals etc)  or is an impossible task now i'm here but bootlops fallback in bootloader mode [00014393] [c0] No dm-verity corruption - using in 'restart' mode.
[00014393] [c0] read_is_device_unlocked() get DeviceStatus 0x1, *out_is_unlocked is 1.
[00014394] [c0] avb_slot_verify() save slot_data to *out_data 
[00014394] [c0] avb_slot_verify() avb slot verify OK!
[00014394] [c0] >>>>>>>>>>>>>>>>  avb_slot_verify result is 5.
[00014394] [c0] >>>>>>>>>>>>>>>>  avb_slot_verify result is ERROR_PUBLIC_KEY_REJECTED.
[00014394] [c0] slot_data[0] is 0x9f74d4e0.
[00014394] [c0] slot_data[1] is 0x0.
[00014394] [c0] avb_check_image() ret is 5.
[00014394] [c0] avb_slot_data[0]->cmdline is androidboot.vbmeta.device=PARTUUID= androidboot.vbmeta.avb_version=1.1 androidboot.vbmeta.device_state=unlocked androidboot.vbmeta.hash_alg=sha256 androidboot.vbmeta.size=47104 androidboot.vbmeta.digest=1e7099b3ee628d5514422a5f32572cbd6f10a3e582549af28fbcc5a01f8c20b8 androidboot.veritymode=enforcing androidboot.veritymode.managed=yes.
[00014394] [c0] g_sprd_vboot_cmdline is androidboot.vbmeta.device=PARTUUID=1.0 androidboot.vbmeta.avb_version=1.1 androidboot.vbmeta.device_state=unlocked androidboot.vbmeta.hash_alg=sha256 androidboot.vbmeta.size=19392 androidboot.vbmeta.digest=a0538b0b30fe79a01c9a9869ff33bd9ede54bce18866c35ecccded4ccf61e678 androidboot.vbmeta.invalidate_on_error=yes androidboot.veritymode=enforcing.
[00014427] [c0] g_sprd_vboot_cmdline is androidboot.vbmeta.device=PARTUUID=1.0 androidboot.vbmeta.avb_version=1.1 androidboot.vbmeta.device_state=unlocked androidboot.vbmeta.hash_alg=sha256 androidboot.vbmeta.size=47104 androidboot.vbmeta.digest=1e7099b3ee628d5514422a5f32572cbd6f10a3e582549af28fbcc5a01f8c20b8 androidboot.veritymode=enforcing androidboot.veritymode.managed=yes.
[00014459] [c0] resolved_hashtree_error_mode = 1.
[00014459] [c0] keyword: = com.android.build.boot.security_patch key_size = 37
[00014459] [c0] WARNNING: LOCK FLAG IS : UNLOCK, SKIP VERIFY!!!
[00014459] [c0] [sprdlk][lcd_printf] enter lcd_printf
[00014464] [c0] uboot_set_root_of_trust succeeded
[00014464] [c0] set_root_of_trust succeeded.
[00014467] [c0] pass_chip_uid_to_tos()... sizeof(blocks)=8
[00014475] [c0] blks:0x00742489 0x02089189, result: 0x0 
[00014479] [c0] lk_set_rpmb_size: rpmb size 16777216
[00014488] [c0] capacity -user: 0x1d1f000000, -boot: 0x400000, -rpmb: 0x1000000
[00014494] [c0] is_wr_mmc_rpmb_key rpmb key has been written
[00014499] [c0] capacity -user: 0x1d1f000000, -boot: 0x400000, -rpmb: 0x1000000
[00014505] [c0] capacity -user: 0x1d1f000000, -boot: 0x400000, -rpmb: 0x1000000
[00014511] [c0] capacity -user: 0x1d1f000000, -boot: 0x400000, -rpmb: 0x1000000
[00014517] [c0] lk_check_rpmb_key get rpmb package, call tos to check rpmb key 
[00014524] [c0] [is_rpmb_erased_check]: country code: 01000100 
[00014529] [c0] trusty_dev_common.c: INFO selected trusty api version: 2 (requested 3)
[00014536] [c0] capacity -user: 0x1d1f000000, -boot: 0x400000, -rpmb: 0x1000000
[00014547] [c0] read  successed
[00014549] [c0] capacity -user: 0x1d1f000000, -boot: 0x400000, -rpmb: 0x1000000
[00014555] [c0] sec_rpmb_read: rpmb block 65529 read successful 
[00014560] [c0] [is_rpmb_erased_check]: rdata[0]:170, rdata[1]:1,rdata[3]:0, rdata[255]:5
[00014569] [c0] read rmiscdata:255
[00014571] [c0] lk_keymint_ipc.c: INFO Initializing Trusty device
[00014576] [c0] trusty_dev_common.c: INFO selected trusty api version: 2 (requested 3)
[00014583] [c0] lk_keymint_ipc.c: INFO Initializing Trusty IPC device
[00014588] [c0] lk_keymint_ipc.c: INFO Initializing LK Keymint IPC service
[00014594] [c0] lk_keymint_ipc.c: INFO Succeeded to send cmd (28) to keymint TA.
[00014600] [c0] lk_keymint_ipc.c: INFO Received 7 byte response
[00014605] [c0] lk_keymint_device.c: INFO Get keymint version 3
[00014609] [c0] s_boot_patch_level = 20240105.
[00014609] [c0] lk_keymint_ipc.c: INFO Initializing LK Keymint IPC service
[00014609] [c0] lk_keymint_ipc.c: INFO Succeeded to send cmd (212992) to keymint TA.
[00014609] [c0] lk_keymint_ipc.c: INFO Received 4 byte response
[00014609] [c0] uboot postload
[00014609] [c0] not in calibration mode
[00014612] [c0] secure_sign_verify consume 11162ms
[00014612] [c0] arch_context_switch: cpu 0, old 0x9f748c90 (secure_sign_verify, pri 24, flags 0x7), new 0x9f7423c0 (sprd_preload, pri 24, flags 0x7)
[00014624] [c0] arch_context_switch: cpu 0, old 0x9f7423c0 (sprd_preload, pri 24, flags 0x7), new 0x9f739010 (bootstrap2, pri 16, flags 0x7)
[00014635] [c0] enter mode normal, boot_reason: Pbint triggered, pwroff_reason: write pwroff
[00014643] [c0] usb_driver_exit entered!
[00014643] [c0] usb_pin_mux config is closed!
[00014646] [c0] waif for core1 process kernel done and powerdown
[00014654] [c2] arch_context_switch: cpu 2, old 0x9f7469a0 (self_monitor, pri 24, flags 0x7), new 0x9f73f5f0 (secondarybootstrap2, pri 16, flags 0x7)
[00014666] [c0] fdt vboot str_len is 338, ret is 0
[00014666] [c2] call the smc to power off the core2
[00014668] [c0] final bootargs: earlycon console=ttyS1,115200n8 loop.max_part=7 loglevel=1 init=/init root=/dev/ram0 rw printk.devkmsg=on swiotlb=1 dummy_hcd.num=0 initcall_debug=1 androidboot.init_fatal_panic=true androidboot.hardware=ums9230_hulk androidboot.dtbo_idx=7 lcd_id=ID13 lcd_name=lcd_hx83108a_boe_mipi_hd lcd_base=9e000000 lcd_size=1600x720 logo_bpix=24 androidboot.upgradeflag=false androidboot.lwfq.type=-1 androidboot.auto.efuse=T612 androidboot.aut [... cut ...]
[00014847] [c0] add bootconfig tailer:size = 00000000, checksum: 0 locate to 0xa2a59e1e!
[00014849] [c0] audio cp boot ok!
[00014849] [c0] [start_linux] start linux at PC:0x80080000, dt_addr:0x9df00000
[00014855] [c0] lk consume time: 13784ms
[00014858] [c0] init log type 0
[00014863] [c0] panic type boot count 2
[00014866] [c0] init log type 1
[00014866] [c0] start type init log success on 2 times
[00014870] [c0] init log type 2
[00014870] [c0] fastboot type boot count 2
[00014873] [c0] init log type 3
[00014873] [c0] download type boot count 2